lexus2k/ssd1306 #3: Can't compile for Atmega328

Yadro, 2017-10-16:
Hi @lexus2k. I tried compile your examples for Atmega328, but I getting a lot of same errors with ` unknown type name 'class'`.
Could you help me?

```
In file included from C:\Users\yadro\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino/Print.h:27:0,
                 from C:\Users\yadro\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino/Stream.h:26,
                 from C:\Users\yadro\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\libraries\Wire\src/Wire.h:26,
                 from E:\MyProjects\Arduino\libraries\ssd1306\src\i2c\ssd1306_i2c.c:151:
C:\Users\yadro\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.20\cores\arduino/Printable.h:25:1: error: unknown type name 'class'
 class Print;
 ^
...
```

lexus2k, 2017-10-16:
Hi @Yadro. I'm looking at the issues. I will back to you as soon, as I find root cause.
For now, could you please rename ssd1306_i2c.c to ssd1306_i2c.cpp, and let me know if this helps.

lexus2k, 2017-10-17:
Hi @Yadro. I fixed the issue in 1.1.3 release. Please, check the latest version

Yadro, 2017-10-17:
Cool, thanks for your work. Result build too small - `Global variables use 320 bytes (15%)` for demo prj. Big thanks!

Yadro, 2017-10-17:
For comparison library from Adafruit used 1568 bytes Global variables.

closed 2017-10-18 by lexus2k