lexus2k/ssd1306 #4: EOL

scargill, 2017-10-26:
Not an issue.. but I've pulled your code as I need an EOL facility... if you add that to yours - I'll scrap mine rather than keep 2.

In my minor change - I've added an optional parameter - see ERASE_EOL

  ssd1306_charF6x8(0, 2, "Test",STYLE_NORMAL,ERASE_EOL);

lexus2k, 2017-10-27:
Hi Peter,

could you please describe scenario of using ERASE_EOL in details?

scargill, 2017-10-27:
Well had I changed the code to globally store X I could have made a separate erase_to_eol() command... as it is, I added it to the text drawing routine...

So simple enough - you have a display - let's say line 2 has "Voltage reading 230.3v" -  next time around - without wanting to wipe the screen, it might say "Voltage reading 99.4v" -  you obviously end up with an extra character at the end - as this string is shorter. You can get around that with a load of formatting but it's a lot easier simply to have an erase to end of line.

scargill, 2017-10-27:
Well, the canvas version appears to work well, but the clearblock function is not available therein so my clear to EOL becomes a little more difficult :-)

lexus2k, 2017-10-29:
Hi Peter
Could you please check the latest committed sources. I've added new API function, which can be suitable for your needs: ssd1306_charF6x8_eol(left, y, ch[], style, right). The last parameter - right - is just what you need, and it can be set also to any value different from last right position on the display.

scargill, 2017-11-02:
Hi thanks for that, not had a chance to check due to travel but this is definitely a worthwhile addition to the code!

closed 2017-11-08 by lexus2k