lexus2k/ssd1306 #6: Fill

scargill, 2017-10-27:
Not QUITE understanding the fill rect function in canvas

  canvas.fillRect(80,20,100,25,1);

That last parameter - I've tried 1, 255, -1  - I cannot fill a rectangle in white on the SSD1306 32 high display...  rectangle and text works just fine

???

scargill, 2017-10-27:
I'll rephrase that - the function  as called above, if called after I've drawn some text stops all updates - my text updates don't happen. If I comment out that fill, everything works...

lexus2k, 2017-10-28:
I fixed the issue with fillRect(). Please, check latest sources.

lexus2k, 2017-10-31:
Let me know if the fix works for you. I checked on Nano with ssd1306 and now filled rectangle looks good.

lexus2k, 2017-11-02:
Any news on the issue?

lexus2k, 2017-11-08:
I'm closing the issue. If you still observe the problem in the latest official release, please, reopen the issue

closed 2017-11-08 by lexus2k