lexus2k/ssd1306 #10: Need better #ifdef for pgmspace.h

tedder, 2017-11-21:
The "#ifdef ESP8266" doesn't cover the ESP32, which makes sense as it isn't an 8266. [Here's one tactic](https://github.com/SodaqMoja/Sodaq_DS3231/issues/5#issue-146403640) that might work more generally. [Here's the same concept listed in the main esp32 lib](https://github.com/espressif/arduino-esp32/issues/508#issuecomment-314968399).

tedder, 2017-11-21:
I forked and started to fix this, but.. there are bigger problems, like stdint, interrupts (SREG, DDR_REG). [Here's the library that has ESP32 and ESP8266 support](https://github.com/squix78/esp8266-oled-ssd1306).

lexus2k, 2017-11-21:
Hi @tedder ,

ssd1306_i2c_embedded.h is not supported for ESP. So, to fix the issue with interrupts, it is needed to update ssd1306_i2c_conf.h (refer to ESP8266 branch). I didn't test my library on ESP32 (because I don't have it), and that's why it is not pointed in the list of supported platforms. Of course, it is possible to update the ssd1306 library, but I need someone to verify.
Why stdint is the problem for ESP32? stdint is standard c-library and is present on all platforms.

I'm glad that you found the library, you can use.

lexus2k, 2017-11-21:
I've submitted changes for ESP32, but didn't check them (the reason is described earlier in this thread)

tedder, 2017-11-21:
Yeah, it seemed like there were enough problems that I gave up.

stdint: had to import the header, uint_8 wasn't recognized.

tedder, 2017-11-22:
I see your changes- I'll pull 'em.

lexus2k, 2017-11-22:
I checked compilation for ESP32. No issues found, see commit  1d5d260.

lexus2k, 2017-11-28:
Hi @tedder,
Any news? Do you still have issues with compilation?

tedder, 2017-11-28:
Hey, sorry. I needed to make more changes. Here they are. stdint is certainly needed in many places, and I guessed on the buffer length. I assume WIRE_SUPPORTED is not set, but I didn't investigate.

```
diff --git a/examples/ssd1306_demo/sova.h b/examples/ssd1306_demo/sova.h
index 905476c..7205265 100644
--- a/examples/ssd1306_demo/sova.h
+++ b/examples/ssd1306_demo/sova.h
@@ -21,11 +21,12 @@
 #define _SOVA_H_
 
 // ----------------------------------------------------------------------------
-#ifdef ESP8266
+#if defined(ESP8266) || defined(ESP32) || defined(ESP31B)
     #include <pgmspace.h>
 #else
     #include <avr/pgmspace.h>
 #endif
+#include "stdint.h"
 
 extern const uint8_t Sova [] PROGMEM;
 
diff --git a/src/font6x8.h b/src/font6x8.h
index 4581cca..26051a0 100644
--- a/src/font6x8.h
+++ b/src/font6x8.h
@@ -16,6 +16,7 @@
 #define FONT6x8_H
 
 // ----------------------------------------------------------------------------
+#include "stdint.h"
 #if defined(ESP8266) || defined(ESP32) || defined(ESP31B)
     #include <pgmspace.h>
 #else
diff --git a/src/i2c/ssd1306_i2c_wire.cpp b/src/i2c/ssd1306_i2c_wire.cpp
index b780c03..0be7ec4 100644
--- a/src/i2c/ssd1306_i2c_wire.cpp
+++ b/src/i2c/ssd1306_i2c_wire.cpp
@@ -66,8 +66,10 @@ void ssd1306_i2cSendByte_Wire(uint8_t data)
     // Do not write too many bytes for standard Wire.h. It may become broken
 #ifdef BUFFER_LENGTH
     if (s_bytesWritten >= (BUFFER_LENGTH >> 1))
-#else
+#elif USI_BUF_SIZE
     if (s_bytesWritten >= (USI_BUF_SIZE -2))
+#else
+    if (s_bytesWritten >= 31)
 #endif
     {
         ssd1306_i2cStop_Wire();
diff --git a/src/i2c/ssd1306_i2c_wire.h b/src/i2c/ssd1306_i2c_wire.h
index 6d7f9c4..582c414 100644
--- a/src/i2c/ssd1306_i2c_wire.h
+++ b/src/i2c/ssd1306_i2c_wire.h
@@ -24,8 +24,8 @@
 #ifndef _SSD1306_I2C_WIRE_H_
 #define _SSD1306_I2C_WIRE_H_
 
-#include <stdint.h>
 #include "ssd1306_i2c_conf.h"
+#include <stdint.h>
 
 #ifdef SSD1306_WIRE_SUPPORTED
 
```

lexus2k, 2017-11-28:
Hi Ted,

The changes, you described, are already present in commit 1d5d260. You can take latest sources from master branch.

tedder, 2017-11-28:
Aha, I was at afb6bf33772baf7bcfd6c029ee126aa6bae740ad. Let me try again.

tedder, 2017-11-28:
Yeah, they compile at least. It'll take me a day or two to test the hardware if you want.

lexus2k, 2017-11-29:
If you have free time to check, it would be nice. I will be able to verify as soon as I find ESP32 module.

tedder, 2017-11-30:
I connected a 128x32 to a DOIT ESP32 kit board. I then opened draw_text, changed the init, and commented out the last two lines since it's a smaller device. The text shows but gets progressively more "broken" later in the line. It looks like a clock issue, maybe?

Here's a pic of how it looks:
https://tedder.me/arduino/doit-32x128-corruption.jpg

I have a handful of other modules on order.

lexus2k, 2017-11-30:
Maybe, it looks like some i2c issue. Do you use PULLUP resistors for i2c lines?
Can you reduce i2c speed to standard 100kbps instead of fast (400kbps).

This is what my 128x32 lcd shows with ESP8266-01 module.
[image]

Updated sketch:
```
#include "ssd1306.h"
#include "i2c/ssd1306_i2c.h"

void setup()
{
    /* Replace the line below with ssd1306_128x32_i2c_init() if you need to use 128x32 display */
    //              scl_gpio, sda_gpio, 0
    ssd1306_i2cInitEx(2,0,0);
//    ssd1306_128x32_i2c_init();
    ssd1306_128x32_init();
    ssd1306_fillScreen(0x00);
    ssd1306_charF6x8(0, 0, "Line 1. text");
    ssd1306_charF6x8(0, 1, "Line 2. Bold text", STYLE_BOLD);
    ssd1306_charF6x8(0, 2, "Line 3. Italic text", STYLE_ITALIC);
//    ssd1306_charF12x16(0, 3, "Line 4. Double size", STYLE_BOLD);
}
```

tedder, 2017-11-30:
Hmm, it's supposed to have internal pullup resistors. I'll investigate that, try your sketch, and add manual resistors if necessary.

tedder, 2017-12-01:
Test results- couldn't figure out how to do the internal pullup. Probably user error. Playing with various ohm/hz combinations:

390Ω + 1,000,000hz: garbage on whole screen
5.6kΩ + 1,000,000hz: nothing
12kΩ + 1,000,000hz: nothing
(didn't bother trying higher)

1MΩ + 100,000hz: garbage/screen doesn't get blanked or take commands
12kΩ + 100,000hz: garbage/screen doesn't get blanked or take commands
5.6kΩ + 100,000hz: dirt on second half of 4th line, can write even to end of line
390Ω + 100,000hz: dirt on second half of 4th line, can write even to end of line

I don't know if that helps. Here's my code: https://gist.github.com/tedder/57ddc04f1c6806f191101552fe87dd18

lexus2k, 2017-12-02:
Initialization in your code looks good. Take into account that ssd1306 controller supports only standard (100kHz) and fast i2c (400kHz) according to datasheet. Due to capacitance of wires, too big pull-up resistor values are not enough, but too small are dangerous, because of big current if line is set to 0. Usually, it is ok to use 4.7kΩ - 10kΩ (in many mass market products they use 4.7kΩ).

I'm surprised that 12kΩ in 100kHz mode didn't work for you. Do you have some sort of logic analyzer tool, like saleae logic analyzer?

lexus2k, 2017-12-04:
Hi,
It seems I fixed issue with ESP32. At least all demos from ESP8266 list work now. The i2c issue on ESP32 is known issue (https://github.com/espressif/arduino-esp32/issues/839), If i2c session lasts for too long, something happens with i2c lines. So, I reduced single i2c session time, and now it works ok.

This is behavior before the fix:
[image]

This is after fix:
[image]

Please, check the latest sources on master.

lexus2k, 2017-12-08:
@tedder,

Did you have a chance to try the fix?
I performed tests for ESP32 and the library works good with my ESP32 dev board.

tedder, 2017-12-08:
ah, didn't see your followup. I'll get to it, certainly this weekend if not sooner. I bet it'll work.

lexus2k, 2017-12-16:
Hi @tedder ,
I'm closing the issue. On my ESP32 the code works.
If you still experience issue with ssd1306 library, please let me know

closed 2017-12-16 by lexus2k

tedder, 2017-12-17:
Hi Aleksei, sorry I dropped the ball on getting back to you. This works completely on my ESP32 test setup. Looks beautiful!