lexus2k/ssd1306 #13: samd21 port

deladriere, 2017-12-06:
I am trying to run this on the Adafruit feather M0 or on the Arduino Zero, but it won't compile
any idea how to get it running?
many thanks

lexus2k, 2017-12-06:
Hi @deladriere ,

Unfortunately I don't any SAMD board by hand. And I'm not sure what compiler definition to use for these boards. So, I fixed compilation for SAMD by adding `defined(ARDUINO_ARCH_SAMD)` blocks.
Download latest master branch and let me know if it works for. Please, be careful, since I didn't publish new release yet. You need to get latest source of ssd1306 library and put them to libraries folder of Arduino IDE.

deladriere, 2017-12-06:
It works ! thanks a lot 
(now I will compare its speed performance compared to the Adafruit ssd1306 library)

lexus2k, 2017-12-06:
Take into account, that Adafruit library always uses buffered output. While ssd1306 library's main mode is direct writing to LCD display. That is, if you need to draw single pixel on the display, the following actions will take place:

**Adafruit**
1. draw pixel in buffer (for example 128x64 LCD)
2. send buffer to LCD (this will require sending of 1024 bytes over i2c)

**ssd1306**
1. draw pixel directly to LCD by sending several bytes to LCD

So, the performance greatly depends on what you need to do in your project.
Of course, ssd1306 library also has some support of buffered output (see NanoCanvas) and some another kind of fast buffers: SpritePool (see loderunner and snowflakes examples).

deladriere, 2017-12-07:
Thank you so much for your help 👍 
What I want to do is "just" write 3 lines of large text and refresh them at high speed to avoid slowing down my main code (a kind of musical instrument, where tempo is critical)

I wrote a test code to compare the 2 libraries :

I am pushing the I2C clock to the max but it still needs 23 ms to redraw the screen.
(here is the code I use [lexus2K SSD1306 I2C text speed test](https://gist.github.com/deladriere/ce9b956744f47b73bd579d335b998461)

The Adafruit code does a little better with 20 ms and no flicker
(here is the code I use [Adafruit SSD1306 I2C text speed test](https://gist.github.com/deladriere/9c5669f359ae4551fb08ffa6b980d181)

Any suggestions welcome!

lexus2k, 2017-12-07:
In the I2C text speed test you wrongly set i2c clock to 2500000:
```
long I2C_clock=2500000;
...
    Wire.setClock(I2C_clock);
```
while in Adafruit text speed to set it to 250kHz.
Remember that ssd1306 controller supports max 400kHz.

deladriere, 2017-12-07:
Ok I will check with an oscilloscope because my sketch works fine until 2700000 before crashing

lexus2k, 2017-12-07:
Hi @deladriere,
I reviewed your tests. You cannot compare too libraries via tests, performing different operations.
Your Adafruit test does all operations in memory and then just draws buffer on the screen (sending 1024 bytes via `display.display()`), while your ssd1306 test doesn't use any buffer: clears controller memory directly over i2c (1024 bytes), and then prints all text again over i2c. Remember that i2c operations are always very slow comparing to SRAM operations. Thus in your tests ssd1306 library does much more work over i2c.

Adafruit library cannot do any graphics directly to OLED without using any memory buffer. So, to do similar tests, you need to use NanoCanvas.

This is the test, I developed for Adafruit and ssd1306 library (it uses buffer for both cases): [speedtest.txt](https://github.com/lexus2k/ssd1306/files/1538565/speedtest.txt) (comment/uncomment `#define ADAFRUIT_TEST`).
**For both libraries I get 47ms results**

But with ssd1306 you can work with LCD without buffer, reducing SRAM memory consumption.

Also, be careful with the speed: ssd1306 datasheet from manufacturer says "i2c Clock Cycle Time  min 2.5us", that means that in i2c mode ssd1306 LCD cannot work faster than 1/2.5us = 400kHz. So, refer to i2c timing characteristics for your LCD device.

deladriere, 2017-12-08:
Thanks !
I have 38 ms with your library but 122 ms with the Adafruit !
I would like to adapt your code to use the bigger fonts like charF12x16 to see if the speed is still good.
How can I do that ?

lexus2k, 2017-12-09:
Please, check commit 9dd141a. I added charF12x16 to NanoCanvas, but didn't check.

deladriere, 2017-12-09:
charF12x16 fonts work, thanks !

lexus2k, 2017-12-11:
You're welcome.
I'm closing the issue since SAMD platform is fixed

closed 2017-12-11 by lexus2k