lexus2k/ssd1306 #15: New fonts

R9YAG, 2018-01-06:
Hi,

Using this library for watch project. Please add more fonts. As I understood I can only add custom fonts not bigger than 6x8, but for watch i defenetly need a bigger one.

lexus2k, 2018-01-06:
Hi @R9YAG,

What platform are you using for the development? Is it AVR, Extensa or ARM?
Yes, now you can add only 6x8 fonts and there is special function to print fonts in double size 12x16.

What font size would you like to have support for?

ssd1306, pcd8544 controllers, and I think, many other monochrome lcd controllers, represent internal RAM byte as 8 vertical pixels. Thus, working with the height of a multiple of 8, makes implementation simple.

lexus2k, 2018-01-11:
Please, try the latest version 1.4.0

lexus2k, 2018-01-25:
There is cool tool [GLCD Font Creator](https://www.mikroe.com/glcd-font-creator). You can create the font you need with it.
Just remember that fonts of ssd1306 library have special format. First 4 bytes of the array should be:
0x00, width_px, height_px, ascii_offset (usually 32).

Other bytes represent chars 1 by 1. Each char should have specially aligned bytes. For example, for char 6x16 the bytes should be: R0C0,R0C1,R0C2,R0C3,R0C4,R0C5,R1C0,R1C1,R1C2,R1C3,R1C4,R1C5. RxCx element is byte, representing eight vertical pixels.

lexus2k, 2018-02-20:
I'm closing this issue, since there have been no any response for a long time.
Let me know, if you need some special font.

closed 2018-02-20 by lexus2k