lexus2k/ssd1306 #23: `%` character flipped

CromFr, 2018-01-29:
Not sure if it is intentional, but the default font for rendering the `%` character is fliped left/right (or top/bottom depending on how your brain works ^^)

However the courier_new_font11x16_digits displays the `%` character correctly

## LCD display type
SSD1306

## Steps to reproduce the issue
```c
#include "ssd1306.h"
#include "font6x8.h"
void setup()
{
    ssd1306_128x64_i2c_init();
    ssd1306_fillScreen(0x00);
    ssd1306_setFixedFont(ssd1306xled_font6x8);
    ssd1306_printFixed  (0,  8, "This % isn't right", STYLE_NORMAL);
}
void loop(){}
```

## Actual behavior
[image]

lexus2k, 2018-01-29:
Hi Charles,

Good finding. Fixed in commit, attached to this issue.

CromFr, 2018-01-29:
Damn you're quick :)
Thanks !

lexus2k, 2018-02-02:
1.4.7 with the % char fix is released.

closed 2018-02-02 by lexus2k