lexus2k/ssd1306 #24: move bitmaps

iigum, 2018-01-31:
Is it possible to move bitmaps? How to implement this? Thank you.

lexus2k, 2018-01-31:
Hi,

Could you please describe your case in details. The way to go depends on platform, you're using for development, and type of OLED/LCD display, you have. For example, ssd1331 has hardware acceleration for moving bitmaps, ssd1306 can do display scrolling.
In general, many if SOC has enough RAM, this can be done via double buffering (i.e., using NanoCanvas of ssd1306 library), but if resources are limited, it is impossible to use double buffering, and another logic needs to be implemented and used.

iigum, 2018-02-01:
Hi Aleksei,
platforms: esp8266 and stm32
displays: ssd1306
I need to move/scrool (up-down-left-right) current bitmap(128x64).
I want to create a menu based on bitmaps and manage it with a mini-joystick.
Scrolling or movement should be as smooth and fast as possible.
That is, there are a several bitmaps and I need to navigate between them. Frame_1 (or canvas_1) Frame_2, etc.

Thank you))
Если вы по русски говорите, быть может я смогу лучше объясниться?))

lexus2k, 2018-02-01:
I speak Russian ;) You can find my email in library.properties file of the library. So, we can discuss the case over email. I would like to keep the discussion on github in English to respect everyone, interested in this project.
**Option 1**
Since you're using ESP8266 and/or STM32 you have enough RAM to use double buffering (128x64 buffer will take 1KiB only). Draw bitmaps as you need, and the output whole buffer to OLED display. SSD1306 datasheet claims 400kHz is supported. That means that theoretically you can reach 400000/(128*64)=48 fps. So, using double buffering is good option for you. I added drawBitmap function to NanoCanvas, here is example of output (with oled initialization):
```
    uint8_t buffer[128*64/8];
    NanoCanvas canvas(128,64, buffer);
    ssd1306_clearScreen();
    canvas.clear();
    for(uint8_t i=0; i<64; i++)
    {
        canvas.clear();
        canvas.drawBitmap(0, (uint8_t)(i-64), 128, 64, Sova);
        canvas.blt(0, 0);
    }
```
On my Atmega it is pretty slow, but, maybe it will be fast on ESP.

**Options 2**
ssd1306 oled has hardware scrolling. I walked through ssd1306 spec, but I didn't find scrolling useful for your use case. If your case is considered to have single bitmap during scrolling, then hardware acceleration will be good for you.

lexus2k, 2018-02-01:
And ssd1306 can communication with host uC via different interfaces:
i2c is the slowest one (400kHz max) while spi is much faster: up to 10MHz

lexus2k, 2018-02-10:
Please, check also commit c275143: I added new function gfx_drawMonoBitmap(). At allows to draw bitmap directly on ssd1306 oled at any y-position

lexus2k, 2018-02-20:
Did the new function help you?

iigum, 2018-02-20:
Oh yeah. Thank you. sorry for the delay.

lexus2k, 2018-02-21:
Ok, thanks.
Let me know if you need anything else.

closed 2018-02-21 by lexus2k