lexus2k/ssd1306 #25: esp32 non-standard i2c pins

zerog2k, 2018-02-02:
is there a way to init esp32 for non standard i2c pins?
I have a dev board with ssd1306 i2c hardwired to pins 4,5 (scl, sda).
I didn't see a way to override this during init, e.g. w/ ssd1306_128x64_i2c_init()
I could fix it by hacking directly on ssd1306_i2c_common.c, but wonder if there is a better way?

zerog2k, 2018-02-02:
i see this note in ssd1306_128x64.h about calling `ssd1306_i2cInitEx()` directly:
```
/**
 * @brief Inits 128x64 OLED display over i2c (based on SSD1306 controller).
 *
 * Inits 128x64 OLED display over i2c (based on SSD1306 controller)
 * This function uses hardcoded pins for i2c communication, depending on your hardware.
 * If you use non-standard pins in your project, please perform call ssd1306_i2cInitEx() and
 * ssd1306_128x64_init().
 */
void         ssd1306_128x64_i2c_init();
```

however, `ssd1306_i2cInitEx()` doesn't seem to be in scope/defined if I just include "ssd1306.h".
Do I need to include other stuff to use custom i2c pins?

zerog2k, 2018-02-02:
ok got it to work with this:
```
#include "ssd1306.h"
#include "i2c/ssd1306_i2c.h"
#define SCL_PIN     4
#define SDA_PIN     5
```
and in setup():
```
    ssd1306_i2cInitEx(SCL_PIN, SDA_PIN, SSD1306_SA);
    ssd1306_128x64_init();
```

lexus2k, 2018-02-02:
Hi,

Good idea. There is one moment: not all i2c implementations allow to change sda/scl pins. Some of i2c implementations have other options: for Linux, this is the number of i2c bus. Actually, I added ssd1306_i2cInitEx, because my first esp8266-01 module has only non-standard 2 pins.
I will add new function, initialization in 2 steps confuses many developers.

lexus2k, 2018-02-20:
Could you please confirm that ssd1306_128x64_i2c_initEx() is what you expect to see in the library?

zerog2k, 2018-02-20:
what would the function signature be of  ssd1306_128x64_i2c_initEx() ?

lexus2k, 2018-02-20:
`void ssd1306_128x64_i2c_initEx(int8_t scl, int8_t sda, int8_t sa);`

zerog2k, 2018-02-20:
thanks! yes with latest lib version (1.4.11) this is working good for me:
`ssd1306_128x64_i2c_initEx(SCL_PIN, SDA_PIN, SSD1306_SA);`

closed 2018-02-20 by zerog2k