lexus2k/ssd1306 #29: Add guide on how to port library to new platform

lexus2k, 2018-02-18:
Add guide on how to port library to new platform

desmond-illumadrive, 2018-06-08:
I am looking at porting to another MCU. Any updates on the porting guide.

lexus2k, 2018-06-09:
Hello, I added some description to hal level. Please, check [template](https://github.com/lexus2k/ssd1306/tree/master/src/ssd1306_hal/template).
But remember that you need to implement i2c or spi level for your platform. Please check [Adding new interface documentation](https://github.com/lexus2k/ssd1306/wiki/Adding-new-interface).

Let me know if you have any questions.

lexus2k, 2018-06-10:
I added some detailed instructions to template. Please, check if those steps are clear for you.

desmond-illumadrive, 2018-06-11:
Looks great thanks.

lexus2k, 2018-06-13:
Now, ESP32 support is added and howto is [updated](https://github.com/lexus2k/ssd1306/wiki/FAQ#how-to-port-library-to-new-platform).
The library can be used on ESP32 dev-kits without Arduino IDE.

lexus2k, 2018-06-17:
I added template file to add new oleds: [oled_template.c](https://github.com/lexus2k/ssd1306/blob/master/src/lcd/oled_template.c).
I hope, it will be also useful