lexus2k/ssd1306 #31: Problem after update

AlexDuino, 2018-02-19:
Please provide the following information:
## library version  > v1.35
## LCD display type SSD1306 - 128 x 64 OLED
## Before update the library (old one v1.35 - working) 
Use Tiny_ssd1306.h and Tiny_ssd1306.cpp and i initialize the library with 
oled.beginI2C(0, 2, 0x3C)

## Expected behavior
Old library v1.35 work with attiny85 without problems
## Actual behavior
New library after 1.35 don't display on OLED..... stay black.

How i can fix....
The example from new library (bigger version then 1.35 ) not working. Black screen
With v1.35 no problems with attiny85, the examples work without problems

thanks

lexus2k, 2018-02-19:
Greetings.

I fixed the issue in 1.4.11 (see commit 28b613c). I'm sorry for that.

lexus2k, 2018-02-20:
Could you please, confirm that library works now?