lexus2k/ssd1306 #33: Araknoid compile fails for Arduino Nano after upgrade from library 1.4.12 to 1.5.0

DaveCasey, 2018-03-27:
Please provide the following information:
## library version
1.5.0
## LCD display type
default
## Steps to reproduce the issue
update library to 1.5.0 using arduino 1.8.3 IDE, then compile araknoid example for arduino nano
## Expected behavior
compiles with library 1.4.12
## Actual behavior
fails with library 1.5.0 with the following:

Arduino: 1.8.3 (Mac OS X), Board: "Arduino Nano, ATmega328P"

Documents/Arduino/libraries/ssd1306/examples/arkanoid/arkanoid.ino: In function 'void drawPlatform()':
arkanoid:211: error: 'ssd1306_setRamBlock' was not declared in this scope
     ssd1306_setRamBlock( pos + LEFT_EDGE + 1, PLATFORM_ROW, platformWidth + PLATFORM_SPEED * 2 );
...

lexus2k, 2018-03-27:
Hello,
Could you please download latest master branch and check compilation?
I released 1.5.1

DaveCasey, 2018-03-27:
It compiles with 1.5.1 now. Thanks!

closed 2018-03-27 by DaveCasey