lexus2k/ssd1306 #34: Architectural improvements

b-desconocido, 2018-03-31:
It is more like a proposal to improve the NanoEngine rather than issue. I thought it might be great to have an ability to pass driver callbacks (via functors, function pointers, tagging/specialization, etc) or driver objects instead of calling "ssd1331_fastDrawBuffer16" or "ssd1306_drawBuffer" inside NanoCanvas implementation. The idea behind tiled rendering is very great, but it isn't very easy to port this lib to the other platform / display controller - it takes way too much unnecessary effort.

lexus2k, 2018-04-01:
Hello, I like your proposal. As I understand you mean to make NanoCanvas1, NanoCanvas8, NanoCanvas16 as templates, to have something likes this: `NanoCanvas1<ssd1306_drawBufferFast>`.
So, using this way, it will be possible to port NanoCanvas and use even different LCD library.
Is this what you meant?

b-desconocido, 2018-04-01:
Not sure what is the best way to pass a blitter function into the NanoCanvas, i'm not a C++ master. It shouldn't be hard to understand for beginners, so maybe templates and some default aliases for predefined controllers will be good enough. It opens up some opportunities for other developers, such as using another display controllers, DMA (requires an ability to switch buffers for the canvas) and color indexing (palettes). The blitter's signature looks exactly the same for me - x, y, buffer and it's length (or width and height), with an exception of monochrome display controllers which mostly are using page/bank mode. I was messing around to use same technic (tiled framebuffer + draw callback) yesterday and was pretty happy with the results - 18ms for drawing and 35ms for copying data to the screen on the Blue Pill + ILI9341 SPI at 36MHz. Should be a bit less with DMA enabled or using indexed color mode.

lexus2k, 2018-04-01:
Well, thank you. You gave me a lof of food for thoughts. I understand your point. 
To implement the idea, you described, first NanoCanvas classes need to be refactored. I made some initial rework, trying to make NanoCanvas1/8/16 still compatible with older library releases, but I didn't figure out yet on how to be with blt functions.

The main points are:
* templating NanoCanvas1/8/16 greatly reduces header size.
* implementation of basic draw operations must be separated from blitter functionality, otherwise the things become too complicated.
* blitter should be implemented as extra functionality for the canvas objects.

Making blitter function as template parameter leads to too complicated code. Another solution is to make blitter function as constructor argument, but this make the compiler to generate more code, which is not good for AVR controllers. And one more idea is to use virtual functions for blitter. I'm still looking for the best solution here.

You can find sketches of new NanoCanvas in https://github.com/lexus2k/ssd1306/tree/1.5_canvas branch.

Regarding speed, I have some local changes for the library, not yet committed, and those changes improve buffer copying by 30% on Atmega328p & ssd1306.
By the way, you're talking about TFT with 36MHz SPI and theoretical minimum, and Atmega328 gives only 8MHz over SPI. Which controller do you use?

b-desconocido, 2018-04-01:
I thought it is possible to make an alias using old canvas names (using current blitter by default) to keep compatibility and simplicity, but i'm not sure if it is the correct way. 
As for the MCU, I'm using Blue Pill (cheap Chinese clone), MCU is STM32F103C8T6, basic specs are the following:
- ARM Cortex M3 running at 72MHz
- 20KiB RAM
- 64 KiB FLASH (although all of these boards in my junk box actually have 128KiB of flash as a sweet bonus)
- 3.3V I/O, some digital pins are 5V tolerant
- SPI, I2C, lot of GPIO, 12-bit ADCs, DMA, USB (not host), etc
- should be compatible with Arduino environment and IDE, although i personally don't like this IDE
- US$ ~2 including delivery
I found this board very powerful, inexpensive, free IDE and tools (CubeMX, TrueSTUDIO or Eclipse, gcc/gdb) and, most importantly, it is easy to debug using ST-Link/V2 (also around US$ 2). The live debugging, like conditional/unconditional breakpoints, call stack, memory/variable live view are crucial for me as for MCU newbie.

lexus2k, 2018-04-02:
Oh, you have powerful controller.
Please, check latest commit to https://github.com/lexus2k/ssd1306/tree/1.5_canvas branch. I;ve added example on how NanoCanvas blitter can be specified via NanoCanvasBase class.
But I still doesn't like implementation, and it seems that the better way is to use virtual functions for the Blitter.

b-desconocido, 2018-04-02:
Oh, that is almost exactly what i had in mind, thanks 😃 I'm currently messing around with templated version of my own crappy canvas, using CoordIntType/Width/Height/PixelFormat/ColorFormat as template parameters and other helper templates. What i'm trying to do is to separate color/pixel memory representation and access patterns (i know it is a waste of time, but good enough to study C++/C++11 and templates instead of reading useless foobar-like examples). It seems like gcc can output some nice binaries with very clever optimizations if it knows enough at the compile time.

lexus2k, 2018-04-03:
Yeah, I agree. Initially I thought that implementing library in C-style is good point for small micro controllers. That's why i2c/spi/intf/lcd parts are implemented in C-style with pointers to selected functions during initialization. 
But modern tool-chains support c++ for all devices, and now I came to conclusion, that I it would be much better to implement low driver logic as template parameter for higher-level APIs. This would produce smaller code in flash and would give higher speed, since controller doesn't need to read pointer first to find out the place to jump to.
The opposite side of templates: they require too much code to be placed to header files and make compilation slower.

lexus2k, 2018-04-03:
Thank you for the good wishes. I looked at the video, and I like it, it looks very nice. I like flagged tiles. They're easier to implement and understand.

Tried to build ssd1306 library for STM32, but Arduino IDE fails to compile C-files, if they call Arduino digital wire functions. They (STM32 support guys) did something completely with STM32 support in Arduino IDE (in some places `#ifdef __cplusplus` macros are missed).

Which toolchain do you use to compile programs for STM32?

b-desconocido, 2018-04-03:
I like flagged tiles too, but sometimes it is faster to render "unaligned" tile and there is much less visible tearing at the borders of the tiles. It depends :D
Haven't received my ATmega328Ps and ESP32 yet, so currently i'm tinkering with these STMs. Didn't try to setup Arduino environment either. I'm using "Atollic ARM Tools" (arm-atollic-eabi-gcc, arm-atollic-eabi-++, default in TrueSTUDIO) and "STM32 HAL" (from STM32CubeMX) for the initialization and low-level access as it correlates better with datasheets. The canvas itself isn't using any low level features, so i could try it when the atmega/esp will arrive.

lexus2k, 2018-04-05:
I added support for stm32duino, and that package has support for Blue Pill as they say.

lexus2k, 2018-04-15:
Sorry for delay, cool. I checked your last video, I'm impressed with the speed.
And finally I decided to use virtual methods for blitter, because they make things simpler. 
I think that latest 1.5_canvas version is the one, I'm going to integrate to master branch to continue development there.

lexus2k, 2018-04-22:
Finally I released 1.6.0, which allows to do customization for NanoCanvas objects

closed 2018-04-22 by lexus2k