lexus2k/ssd1306 #35: Screen rotation feature

orcwarrior, 2018-04-02:
Hello! 
I've just encountered yours library, and I really like the features, the one thing that one might find lacking is screen rotation. Actually, if whole screen (with coords) rotation would be cumbersome to do, maybe just canvas rotating would be easier to implement it would still help alot. And by rotation I meant just simply 90deg rotations, nothing too much fancy ofc.

Greetings.

lexus2k, 2018-04-02:
Hello, what type of LCD do you use?
Are you interested in this feature for NanoCanvas or for direct draw API (without double buffering)?

orcwarrior, 2018-04-02:
I currently using SSD1306 128x32, but I have some 128x64 laying around too. Well I think the easiest way to use rotation (from lib user point of view) it would be to setup by some initial function, where rotation would swap axis so 128x32 becomes 32x128. I really dunno if that would be alot of work to make it. But if it is, then I was guessing than rotation of NanoCanvas would be easier to implement. Thanks for you rapid response, I really appreciate effort you put in this project.

lexus2k, 2018-04-02:
Unfortunately, ssd1306 controller doesn't have hardware feature for rotation, it allows only to flip image in horizontal or vertical direction. If to add rotation feature I see several ways:
* Implementing rotation feature for existing direct draw api functions, (but this would increase the code size twice or more) (binary code size consuming feature).
* Adding new set of direct draw functions to draw in 90-degree mode. This will increase the library source code ~ 1.5 times, but if to use only new functions, the code will fit the low-memory controllers. (development time consuming feature)
* Another way out is to add BLT function to NanoCanvas to rotate the image. This is the cheapest way, but I expect here a lot of cpu consumption (~20 times slower),  because the BLT function needs to take 8 bits from 8 different bytes for single byte calculation to send  to ssd1306 controller (cpu speed consuming feature).

What uC do you use in your project? Is it fast enough to run slow BLT operations?

orcwarrior, 2018-04-02:
So as I supect its not that easy to make it nice & clean solution from user of lib pov, without ruining library small footprint and ability to run on attiny.
But the last way would atlast give an ability to rotate, and it still should run not that bad on arduino nano (which I use) I guess?

lexus2k, 2018-04-03:
If you have I2C oled, then I2C speed is too slow, and it is OK to do rotation during NanoCanvas rendering on display. As for SPI oled, ssd1306 can work at 8MHz, and maximum reached communication speed on Atmega328p is only 3Mbps, what means that Atmega328p already cannot provide reasonable speed already. Rotation operation will slow down communication SPI speed to i2c speed.
I expect that you will be able to refresh OLED content from NanoCanvas ~ 10 times per second. Is it enough for you?

The second way, I described earlier, is the fastest one (Adding new set of functions). 
Do you need to do rotation in run-time, or do you just want to use display in 64x128 mode? If you need to use oled in 64x128 you can just swap x & y coordinates and rotate all bitmaps, you're using.

lexus2k, 2018-04-05:
Hello,
Did you try the assumption, I pointed above?

orcwarrior, 2018-04-08:
Hi,
sorry for late response, unfortunatelly I don't have much time for hobby projects now.
I'm not everyday low level c++ but as I get you simply swapped bits properly with rotation, I would write some function for that In my case since performance isn't most important thing in project (btw: lcd is i2s). For current project, display will only shows sensor values, refreshed in big intervals I think I'll pick other lib I found at which as docs claims It's possible to swap screen orientation.
But I still looking for use of your lib when Ill work on another project - fft visualiser on SPI based lcd used in normal orientation this time ;)
Thanks for you time in explaining possiblities.

lexus2k, 2018-05-26:
By adding support of Adafruit GFX library to ssd1306 library, now rotation feature is available for monochrome displays if to use AdafruitCanvas1.
I added sketch, showing how to use Adafruit GFX library with ssd1306 ([examples/double_buffering/mono_adafruit)](https://github.com/lexus2k/ssd1306/tree/master/examples/double_buffering/mono_adafruit)

lexus2k, 2018-05-28:
There is no rotation feature is expected for direct draw API.
Regarding double-buffered output, you can use AdafruitCanvas1 to solve the issue.

closed 2018-05-28 by lexus2k