lexus2k/ssd1306 #36: Instructions for running demo code on ATmega88PA

kblicharski, 2018-04-23:
I see that you have Atmega328 and Atmega168 as supported devices. I'm pretty new to AVR development, and I was wondering if it would be feasible to run this code for the ATmega88PA without significant alteration. I apologize if this is a really dumb question, and I'm looking forward to your reply.

lexus2k, 2018-04-23:
Actually, it should be possible to run demo code on Atmega88 too. The main difference between all these microcontrollers is flash size and RAM.
Atmega328p has 32KiB flash, Atmega168p 16KiB, and Atmega88 8KiB. If you setup your Arduino IDE  correctly, then you will easily compile sketches from the library for Atmega88. Maybe use will need to cut off some demo code to fit the flash size (it depends on the demo sketch, you want to try on Atmega88)

Some instructions to setup Atmega88 board configuration can be found here.
http://forum.arduino.cc/index.php?topic=376918.0

If you get compilation errors, please let me know (and display type you have)

lexus2k, 2018-04-28:
Did you have a chance to check compilation for Atmega88? 
The command below works, but I have no Atmega88 hardware:
> ./build_and_run.sh -p avr -m atmega88p direct_draw/draw_text

lexus2k, 2018-05-03:
Hello,
any news?

kblicharski, 2018-05-03:
Hi Aleksei, we managed to get it to work for our embedded systems project. Thank you for the help!

closed 2018-05-03 by kblicharski