lexus2k/ssd1306 #37: Plain AVR/C/Atmel Studio for ATMega88PA

hhpatel09, 2018-04-24:
I was wondering how if possible to your this diver for plain AVR and the ATMega88PA. Basically, the thing that I am confused on is I don't understand avr-gcc. I don't understand how to build this and how to build it in Atmel Studio. Sorry, I'm kind of new to all of this. Thanks!

lexus2k, 2018-04-24:
Could you please try the commands:
> get clone https://github.com/lexus2k/ssd1306<br>
> cd ssd1306/tools<br>
> ./build_and_run.sh -p avr -m atmega88p direct_draw/draw_text

This will generate ../bld/direct_draw/draw_text.hex, ../bld/direct_draw/draw_text.out and ../bld/direct_draw/draw_text.bin for your controller.

hhpatel09, 2018-04-27:
I am using the ssd1306

lexus2k, 2018-04-28:
Did you have a chance to try build commands?

hhpatel09, 2018-04-28:
Ya I did try and I am getting errors like: recipe for target 'ssd1306_i2c_twi.o' failed.
This error is for the following: ssd1306_i2c_twi.o, ssd1306_i2c_embedded.o, ssd1306_i2c_linux.o, and ssd1306_i2c_common.o
I am getting this in the Makefile that is auto generated by Atmel Studio.

In addition to this, in all of these files there is a library that is included but it is no where to be found and I am getting errors when I am trying to build. Here is the include: #include "hal/io.h"

Thanks,
Harsh

lexus2k, 2018-04-28:
"hal/io.h" header file can be found right in the library: ssd1306/src/hal/io.h
Could you provide Makefile generated by build system. I believe, there is an issue with include directories list in Atmel Studio IDE project.

hhpatel09, 2018-04-28:
I had to upload the makefile as a txt but it is supposed to be .avr or .dms
Also the hal/io.h library I did not include because I am using i2c. I was wonder if I need to include the entire src folder including all of the files that appear right inside the folder and the other ones (ig. i2c, sci, hal) as well.
Thanks.
[Makefile.txt](https://github.com/lexus2k/ssd1306/files/1957912/Makefile.txt)

lexus2k, 2018-04-29:
First of all in the Makefile you provided, I see only few sources from ssd1306 library:
```
C_SRCS +=  \
../main.c \
../ssd1306_i2c_common.c \
../ssd1306_i2c_embedded.c \
../ssd1306_i2c_linux.c \
../ssd1306_i2c_twi.c
```
It will not work this way, the library needs all files for compilation.
Next in the rule, generating object files, I do not observe path to root library sources:
```
./%.o: .././%.c
	@echo Building file: $<
	@echo Invoking: AVR/GNU C Compiler : 5.4.0
	$(QUOTE)C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-gcc.exe$(QUOTE)  -x c -funsigned-char -funsigned-bitfields -DDEBUG  -I"C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.132\include"  -O1 -ffunction-sections -fdata-sections -fpack-struct -fshort-enums -g2 -Wall -mmcu=atmega88pa -B "C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.132\gcc\dev\atmega88pa" -c -std=gnu99 -MD -MP -MF "$(@:%.o=%.d)" -MT"$(@:%.o=%.d)" -MT"$(@:%.o=%.o)"   -o "$@" "$<" 
```
There should be something like `-I"C:\Users\User\Path_to_the_library\ssd1306\src"`

Unfortunately I don't have Atmel Studio, because it runs only in Windows (my laptop has Linux on board), you can try to read some tips on how to add include path to the project.

lexus2k, 2018-05-03:
Did you try to add all required files to your project?

lexus2k, 2018-05-28:
I'm closing this issue, because it is IDE configuration related, but not the library itself.
If you have any questions, let me know

closed 2018-05-28 by lexus2k