lexus2k/ssd1306 #38: Custom width canvas causes rows to shift left or right

JGoodHub, 2018-05-04:
Version 1.6.1
SSD1306 128x64 OLED

This is the section of code I've been having trouble with, according to the docs I should be able to create a NanoCanvas with any width so long as the height is divisible by 8, by from what I can tell any width value that isn't a power of two creates this shifting effect in the rows of bytes, the below code works fine if the width is something like 32 or 64, but anything in-between e.g. 48 and it bugs out like in the pictures below.

```
#include <ssd1306.h>
#include <nano_gfx.h>

uint8_t buffer[48*64/8];
NanoCanvas canvas(48, 64, buffer);

void setup() {
  Serial.begin(9600);
  
  ssd1306_128x64_i2c_init();
  ssd1306_clearScreen();

  canvas.drawRect(5, 0, 30, 63);  
  canvas.blt(0, 0); 
}
void loop() {}
```

This is an example of what I mean, the complete rectangle was does with a width of 64, the jagged one a width of 48.

[Images](https://imgur.com/a/wag9Yie)

lexus2k, 2018-05-05:
Hello @JGoodHub.

Thank you for the finding. Actually this is lack in documentation. Width value for NanoCanvas in 1.6.1 should be power of 2.
For now I do not support `NanoCanvas` class, defined in `nano_gfx.h`, and going to remove it soon. You can use `NanoCanvas1`, accessible via `nano_engine.h` header. Earlier, to speed-up code execution on AVR Attiny controllers and to avoid of using multiply operation (attiny controllers do not have HW multiplication support), monochrome canvas classes were developed to work with fixed width values, which are power of 2. This requirement is needed for Atmega, ESP controllers, since they have more flash, more rich hardware support, so I prepared 4fce244 changes for you (click the link to find the changes).
Could you please, download latest version of ssd1306 library from master branch and check if this is what you're really need?

JGoodHub, 2018-05-05:
Brilliant, just tested it on a arduino pro mini and it works perfectly, tried with with multiple width values  and the rectangle stayed a constant shape with no layer shifting, thanks for the quick fix.

Also just wanted to thank you for making the library available in the first place, way better than any other out there, especially for game dev.

lexus2k, 2018-05-07:
Thank you for the good words.
I released updated library (v1.6.2)

closed 2018-05-11 by lexus2k