lexus2k/ssd1306 #39: ATtiny84 support ?

classic-audio, 2018-05-19:
Please provide the following information:
## library version ssd1306
## OLED display type ssd1306
## Steps to reproduce the issue - 
## Expected behavior - load the sketch to ATtiny84 
## Actual behavior - errors - no loading
The sketch / drivers work with ATtiny85 - really great. But I need more I/O's and would like to use the ATtiny 84.
What changes do I need to do?

lexus2k, 2018-05-20:
Hello,

you didn't tell me what type of communication you need (i2c or spi), but I suggest that AttinyX4 series has similar hardware modules as AttinyX5 series. So, I committed update for the library, which enables the same options for Attiny84 as for Attiny85: e147530.
Please, check latest master branch and let me know. Unfortunately, I don't have Attiny84 hardware to verify.

classic-audio, 2018-05-20:
Hello, thank you vey much for a fast reply.
I am using i2c.
I have downloaded the opdated library and loade my sketch to ATtiny84.
Now the load process is running without any issues, but the diplay is "black".
I loaded a sketch with different drivers and all is working .
I am using the "recommended" ATtiny connections for the display SDA -> pin 7 and SCL -> pin9

lexus2k, 2018-05-20:
Could you please send me your sketch or part of sketch code, which does library initialization.

classic-audio, 2018-05-20:
Hello 

 

This is the sketch I am working on and would like to use with ATtiny84

 

   Venlig hilsen / Regards

   Frede Kristensen

   Classic Audio

 

| Sønderskovvej 7 | Venø | 7600 Struer | Danmark | Phone: +45 97 86 80 84 | CVR: 21 99 19 96 |

classic-audio, 2018-05-20:
[ATtinySketch.pdf](https://github.com/lexus2k/ssd1306/files/2020132/ATtinySketch.pdf)
I am not shure if the attached file to my mail was availble for you - but here is a PDF version

lexus2k, 2018-05-20:
Thanks. 
Well, when they are talking about "recommended" connections for i2c, first of all they mean physical pin numbers: SDA(7), SCL(9). But physical pin numbers is not the same as logical - according to the picture below the logical pins should be SDA(6) and SCL(4).
[image]

Software implementation of SSD1306 library doesn't use recommended pins by default, because, software implementation can work on any pins. And by default, it uses SCL(5) and SDA(4).
Try to use different initialization function like in example below:
```.cpp
void setup()
{
    ssd1306_128x64_i2c_initEx(4, 6, 0);
    ssd1306_clearScreen();
    txt();
}
```

I will update library to use recommended pins on Attiny84 by default.

classic-audio, 2018-05-20:
Thank you very much,
Now it works.
SCL to Pin 8
SDA to Pin 9

And ..... no problem for me - I did use pin according to recommended and the othe earlier driver I have been using. And I could have asked which pin you prefer.
So you dont need ot change anything for me. I have a working setup now.

I really like you drivers, because of the possibilities also for the small versions of Arduino.
And it is possible to develop using the Uno which I am using and later load to the ATtiny.

Thank you very much again - have a nice day.

classic-audio, 2018-05-20:
............ and by using your suggestion 
void setup()
{
    ssd1306_128x64_i2c_initEx(4, 6, 0);
    ssd1306_clearScreen();
    txt();
}
the suggested pin works.
SDA to pin 7
SCL to pin 9

So everything works just great.

lexus2k, 2018-05-20:
There is helper class Ssd1306Console. So, you could try to write the following code (do not forget to include `ssd1331_api.h` and `ssd1306_console.h`):
```.cpp
#include "ssd1306.h"
#include "ssd1331_api.h"
#include "ssd1306_console.h"

Ssd1306Console  console;

void setup()
{
    /* Replace the line below with the display initialization function, you want to use */
    ssd1306_128x64_i2c_init();
    ssd1306_clearScreen();
    /* Set font to use with console */
    ssd1306_setFixedFont(ssd1306xled_font6x8);
}

void loop()
{
    int voltage = 10;
    ssd1331_setCursor8(0,8);
    /* Here use any methods, provided by Arduino Print class */
    console.print( "Voltage: " );
    console.print( voltage );
    delay(500);
}
```

classic-audio, 2018-05-21:
Thank you very much.
I will try that of course. That means I msut cahnge to another display with the SSD1331 driver.
------------------------------------------
I now have a complete - working project (using the previus metode) using 90% of program space of the ATtiny85.
The project is a Vrms voltage monitor for an audio test system.
Everything works just great.
Again thank you very much for a great driver and fast help.

lexus2k, 2018-05-21:
You don't need to switch to another display driver. Just try the example with ssd1306 display ;)
ssd1331_setCursor8 is just API name. This function will work with all displays.

classic-audio, 2018-05-21:
Great - thank you.
Now I see your point.

lexus2k, 2018-05-21:
You're welcome :)

lexus2k, 2018-05-21:
In latest version on master branch I've added setCursor method to Ssd1306Console, so, you can write:
```.cpp
void loop()
{
    int voltage = 10;
    /* Here use any methods, provided by Arduino Print class */
    console.setCursor(0,8);
    console.print( "Voltage: " );
    console.print( voltage );
    delay(500);
}
```

classic-audio, 2018-05-22:
This is a great improvement which is more easy to use because this is similar with other drivers and serial monitor. 
Thanks.

lexus2k, 2018-05-22:
More improvements to follow. Let me know if something works wrong in console object.

lexus2k, 2018-05-31:
1.7.0 with ATtiny84 support is released

closed 2018-05-31 by lexus2k