lexus2k/ssd1306 #41: compile error esp8266

sorriso93, 2018-06-15:
I'm trying to compile the ssd1351_demo but I get this error:

 #include "hal/io.h"
                    ^
compilation terminated.

Edited usersettings.h, but no way. 
I installed the library both from ide and manually, same result...

lexus2k, 2018-06-16:
Hello,

Which library version are you using? 1.7.1 doesn't have "hal/io.h" header.
It would be nice if you provide detailed compilation log.

lexus2k, 2018-06-16:
Just checked right now compilation for Generic ESP8266 module from Arduino IDE and it's OK:

> 
> Using library ssd1306 at version 1.7.1 in folder: /home/alex/Arduino/libraries/ssd1306 
> Using library Wire at version 1.0 in folder: /home/alex/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/Wire 
> Using library SPI at version 1.0 in folder: /home/alex/.arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/SPI 
> Sketch uses 261864 bytes (52%) of program storage space. Maximum is 499696 bytes.
> Global variables use 32772 bytes (40%) of dynamic memory, leaving 49148 bytes for local variables. Maximum is 81920 bytes.

lexus2k, 2018-06-17:
Did you find the root cause of reported problem?

sorriso93, 2018-06-17:
I’m finishing another project, I’ll get back to this in the next days and update you thks

sorriso93, 2018-06-20:
Hello
this is the log, I'll post also my usersettings.h

Arduino: 1.8.5 (Mac OS X), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, All Flash Contents, 230400"

/Applications/Sviluppo/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Sviluppo/Arduino.app/Contents/Java/hardware -hardware /Users/yoghibear/Library/Arduino15/packages -tools /Applications/Sviluppo/Arduino.app/Contents/Java/tools-builder -tools /Applications/Sviluppo/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/yoghibear/Library/Arduino15/packages -built-in-libraries /Applications/Sviluppo/Arduino.app/Contents/Java/libraries -libraries /Users/yoghibear/Documents/Arduino/libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=all,UploadSpeed=230400 -ide-version=10805 -build-path /var/folders/p0/tylmjpjn57z_7q5tq3q4mwmm0000gq/T/arduino_build_908173 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/esptool/0.4.13 -verbose /Users/yoghibear/Library/Mobile Documents/com~apple~CloudDocs/Privato/progetti casa/progr esp2866/03_library_examples/01_displays/03_examples_ssd1306_1331_1351_il9163/demos/ssd1351_demo/ssd1351_demo.ino
/Applications/Sviluppo/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Sviluppo/Arduino.app/Contents/Java/hardware -hardware /Users/yoghibear/Library/Arduino15/packages -tools /Applications/Sviluppo/Arduino.app/Contents/Java/tools-builder -tools /Applications/Sviluppo/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/yoghibear/Library/Arduino15/packages -built-in-libraries /Applications/Sviluppo/Arduino.app/Contents/Java/libraries -libraries /Users/yoghibear/Documents/Arduino/libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=all,UploadSpeed=230400 -ide-version=10805 -build-path /var/folders/p0/tylmjpjn57z_7q5tq3q4mwmm0000gq/T/arduino_build_908173 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/esptool/0.4.13 -verbose /Users/yoghibear/Library/Mobile Documents/com~apple~CloudDocs/Privato/progetti casa/progr esp2866/03_library_examples/01_displays/03_examples_ssd1306_1331_1351_il9163/demos/ssd1351_demo/ssd1351_demo.ino
Using board 'd1_mini' from platform in folder: /Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1
Using core 'esp8266' from platform in folder: /Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1
WARNING: Spurious .circleci folder in 'Homie' library
Detecting libraries used...
"/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/var/folders/p0/tylmjpjn57z_7q5tq3q4mwmm0000gq/T/arduino_build_908173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"   -DESP8266 "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/variants/d1_mini" "/var/folders/p0/tylmjpjn57z_7q5tq3q4mwmm0000gq/T/arduino_build_908173/sketch/ssd1351_demo.ino.cpp" -o "/dev/null"
"/Users/yoghibear/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/Users/yoghibear/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/var/folders/p0/tylmjpjn57z_7q5tq3q4mwmm0000gq/T/arduino_build_908173/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"   -DESP8266 "-I/Users/yoghibear/Library/Ardu [... cut ...]
In file included from /Users/yoghibear/Library/Mobile Documents/com~apple~CloudDocs/Privato/progetti casa/progr esp2866/03_library_examples/01_displays/03_examples_ssd1306_1331_1351_il9163/demos/ssd1351_demo/ssd1351_demo.ino:33:0:
/var/folders/p0/tylmjpjn57z_7q5tq3q4mwmm0000gq/T/arduino_build_908173/sketch/sova.h:28:20: fatal error: hal/io.h: No such file or directory
 #include "hal/io.h"
                    ^
compilation terminated.
Using library ssd1306 at version 1.7.1 in folder: /Users/yoghibear/Documents/Arduino/libraries/ssd1306 
exit status 1
Error compiling for board WeMos D1 R2 & mini.

sorriso93, 2018-06-20:
... please ignore me...
Downloaded again the examples and now it compiles with no problem... seems solved although don't know how... this evening I will try to upload the sketch to the wemos d1... thks for now

closed 2018-06-20 by sorriso93

lexus2k, 2018-06-21:
Ok. Thank you for the update.