lexus2k/ssd1306 #42: Error Error compiling for oled 1306 in arduino and AtTiny85

wataruland, 2018-07-09:
I am trying to compile the ssd1306_demo from the Arduino 1.8.5 IDE version, with the library version 1.7.2 but I get this error with AtTiny85:

libraries \ ssd1306 \ ssd1306_8bit.c.o (symbol from plugin): In function `ssd1331_setColor ':

(.text + 0x0): multiple definition of `ssd1331_drawBufferFast16 '

libraries \ ssd1306 \ ssd1306_16bit.c.o (symbol from plugin) :(. text + 0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Compiling error for the ATtiny25 / 45/85 card.

Or with an Arduino UNO:

Archiving built core (caching) in: C: \ Users \ wataru \ AppData \ Local \ Temp \ arduino_cache_660515 \ core \ core_arduino_avr_one_f76447c3c6205e5ecc0585c82388c111.a
libraries \ ssd1306 \ ssd1306_8bit.c.o (symbol from plugin): In function `ssd1331_setColor ':

(.text + 0x0): multiple definition of `ssd1331_drawBufferFast16 '

libraries \ ssd1306 \ ssd1306_16bit.c.o (symbol from plugin) :(. text + 0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for the Arduino / Genuino Uno card.

I installed the library both from ide and manually, same result ...,
and I do not know how to continue, or if you want to change something in the configuration files.

lexus2k, 2018-07-09:
Hello,

I'm sorry for the bug. I fixed the issue right now,
can you download the library from master branch and verify?
If you prefer to use IDE manager new version will appear later, and as quick fix you can switch to 1.7.1 version.

Thank you

wataruland, 2018-07-09:
First of all thank you very much, I have done what you told me and it works perfectly, also I have seen that you can change the I2C pines with the call ssd1306_128x64_i2c_initEx (SCL, SDA, SA) and as I say perfect. Again, thank you very much: D

lexus2k, 2018-07-10:
You're welcome.

closed 2018-07-10 by lexus2k