lexus2k/ssd1306 #43: problem with ssd1306 i2c rst pin

aklyk, 2018-07-20:
library version 1.7.3

I have problem to run my ssd1306 with i2c on arduino nano. It runs only with libraries dat have a rst setting.

Also. may be I have crooked hands coz im new in arduino

With Adafruit and OLED_i2c libraries display works well

lexus2k, 2018-07-20:
Hello, could you please try to use 'ssd1306_resetController(reset_pin, 50);' before i2c initialization?
reset_pin is pin number, connected to RST of oled display.
It would nice also, if you provide picture with wire connections to your lcd.

lexus2k, 2018-07-22:
@aklyk , I believe the issue here is that your ssd1306 display needs some time after power up before it will be able to accept commands. 
I walked through ssd1306 datasheet and didn't find any timings except one section, that claims ssd1306 controller needs 100ms on reset before display content becomes available. 
Could you please check that and insert delay(100) instead of ssd1306_resetController call?

lexus2k, 2018-07-29:
@aklyk ,
I think that the issue here doesn't relate to the library itself, but to used hardware in the project design.
So, `delay()` right after power up can help to solve the issues, like yours.

I'm closing ticket.

closed 2018-07-29 by lexus2k