lexus2k/ssd1306 #45: Umlaut support

kalsan, 2018-08-03:
First of all,I'm extremely impressed by the tiny size of the library! It's exactly what I was dreaming of.

I'm currently building a cat feeding machine for a german speaking family, so I'm looking for a way to display the so-called "Umlaute" (äöü). This is analog to the french accents (éàè).

Is there some Umlaut support in ssd1306?

lexus2k, 2018-08-04:
Hi Sandro,

At present ssd1306 library uses plain representation of fonts. I mean, it takes ascii code of the character and then calculates character index in the font table. 
In general, it is possible to support utf8, this will increase code size slightly, and it needs creation new fonts for Umlaute.
I see 2 possible ways to support unicodes in the library:
* first way is to add decoding utf8 to drawText functions, but subtract some offset (encoding number) from the unicode and using that index, calculate character offset in the used font. This way is not correct, but it is simple.
* next way is to update font format and add support for unicode characters. This way requires some tools to be developed (maybe python scripts) to allow users to create unicode fonts.

So, let me know, what you think of.

lexus2k, 2018-08-05:
I added test implementation of utf-8 support, and you may look at `ssd1306xled_font6x8` to figure out, how unicode chars can be encoded.

kalsan, 2018-08-07:
Hi @lexus2k and thank you very much for your fast reaction! The letters "äöü" now work out of the box with the new library :-)

I did some testing and these are the results: The code:

```
#include <Arduino.h>
#include "ssd1306.h"
#include "nano_gfx.h"

SAppMenu menu;

const char *const menuItems[] =
{
    "Umlaut Test",
    "äöü",
    "éàè",
    "ÄÖÜ",
    "ÉÀÈ",
};

void setup() {
  ssd1306_setFixedFont(ssd1306xled_font6x8);
  ssd1306_128x64_i2c_init();
  ssd1306_fillScreen( 0x00 );
  ssd1306_createMenu( &menu, menuItems, sizeof(menuItems) / sizeof(char *) );
  ssd1306_showMenu( &menu );
}

void loop() {
  ssd1306_fillScreen( 0x00 );
  ssd1306_showMenu(&menu);
  delay(500);
  ssd1306_menuDown(&menu);
  ssd1306_updateMenu(&menu);
}

```

Compiled with PlatformIO on an UTF-8 Linux system yields to:
[image]

This means that your library now fully supports non-capital German umlauts without the need to add weird escape sequences like one must do for the Adafruit library.

Capital umlauts as well as the French accents do not appear to work yet with my current configuration.

closed 2018-08-07 by kalsan

lexus2k, 2018-08-07:
Hi Sandro,

You're welcome. I'm going to improve library API.
As for the font, I didn't add full support of all letters. It will take some time (as I draw font manually), and increase font size in flash of your attiny85.
So, the best way is to add only those unicode chars, you need.
Let me know, what you need for your project.

kalsan, 2018-08-08:
Hi Aleksei

For this project (targetting a German-speaking person), äöü are sufficient. However it might be an advantage for your library in general to have the most common accents included. Here are a few tips:

- Most Latin languages like French, Italian and Spanish are mostly covered with éàèëòôóñîíïúùûç
- Most Latin languages drop the accent on capital letters, so É etc. are not crucial
- In old-style German, capital ÄÖÜ are replaced by Ae Oe and Ue, making these non-crucial as well

Hope this helps :-)

kalsan, 2018-08-08:
One idea would be to have `CONFIG_SSD1306_UNICODE_GERMAN` etc. per language that then enables the characters that are needed for a specific language.

lexus2k, 2018-08-12:
Hi Sandro,

thank you for the tips. I decided to go with 2 unicode fonts, which user can set: primary and secondary. For example, primary unicode font can be used for standard ascii characters fonts. And secondary can be used for national language.
```cpp
    ssd1306_setFixedFont(ssd1306xled_font6x8);
    ssd1306_setUnicodeTable(ssd1306xled_font6x8_German);
    ssd1306_printFixed (0,  8, u8"Line 1.äöü", STYLE_NORMAL);
```

Thus, end user can choose the way, how to use limited memory resources.

Thank you once more.

kalsan, 2018-08-12:
That's a very elegant solution! Thank you so much for your quality work!