lexus2k/ssd1306 #46: Custom Font definition

sorriso93, 2018-08-23:
Hello
I would try to use a custom font using the font generator by squid, http://oleddisplay.squix.ch/#/home.
What kind of library version should I choose to generate the header file? (> 3.0, adafruit, < 3.0)

many thanks

lexus2k, 2018-08-23:
Hello
I checked that link. That is really good font generator. Unfortunately it generates fonts with dynamic chars width, and ssd1306 can use only fixed width fonts. So, I can learn the font format over the link, and get back to you with library update.
But it will take some time.

Let me know, what you think.

sorriso93, 2018-08-23:
Well if you think can be useful to others to have dynamic chars width for me is ok.
As alternative do you know a similar font generator for fixes width fonts?
Many thanks

lexus2k, 2018-08-24:
Please check GLCD Font creator tool: https://github.com/lexus2k/ssd1306/wiki/How-to-create-new-font-for-the-library

lexus2k, 2018-08-30:
Hello, I added generator, written in python, for new font format. It is possible now to generate fonts from TTF files.
Demo draw_unicode_text demonstrates how new fonts work.

sorriso93, 2018-08-30:
Many thanks, as soon as I have time I'll look into it.

sorriso93, 2018-09-10:
HI 
tried to test the font generator... sorry for the stupid question, on osx which version of pyton should I install and how I can run the font generator (an example of command line and parameters would be great)
many thanks

lexus2k, 2018-09-11:
Hello,

example of font generating from ttf file:
> python ./fontgenerator.py --ttf FreeMono.ttf -s 7 -d -f new > my_font.h

then `#include "my_font.h"` to your application and set the font with:
```.cpp
ssd1306_setFreeFont(free_FreeMono8x8);
```

Here you can find details instructions: https://github.com/lexus2k/ssd1306/wiki/How-to-create-new-font-for-the-library

sorriso93, 2018-09-11:
Should I change the fontgenerator.py to see the freetype module? below a log of the commands, or in alternative change/add relative path somewhere in my osx configuration (must admit I don't know where...)
thks

MaxMacBookAir:tools yoghibear$ python fontgenerator.py --ttf "Fonts/Arial Black.ttf" -f old > ssd1306font.h
Traceback (most recent call last):
  File "fontgenerator.py", line 137, in <module>
    from modules import ttfsource
  File "/Users/yoghibear/Library/Mobile Documents/com~apple~CloudDocs/Privato/progetti casa/progr esp2866/03_library_examples/01_displays/04_examples_ssd1306_1331_1351_il9163/tools/modules/ttfsource.py", line 33, in <module>
    import freetype
ImportError: No module named freetype
MaxMacBookAir:tools yoghibear$ freetype-config --cflags
-I/usr/local/opt/freetype/include/freetype2
MaxMacBookAir:tools yoghibear$

lexus2k, 2018-09-12:
Did you install freetype python module?
You need to use python 2.7 and install freetype python module via pip command:
> sudo pip install freetype-py

I don't have osx, so you need to search over Internet, how to install python modules on your system

sorriso93, 2018-09-12:
.... I thought I installed it... you are right, now I generated the .h file, this night will try the custom font... many thks I let you know when my project will be finished.

lexus2k, 2018-09-13:
ok, I will be waiting for your feedback.

sorriso93, 2018-09-13:
Hello, below the code. I use ST3551 on SPI (maybe this can be the problem)... Anyway, no error on compiling and execution but the screen remains totally black...
Thanks

#include "ssd1306.h"

extern const uint8_t Raleway_Black13x11[] PROGMEM;
const uint8_t Raleway_Black13x11[] PROGMEM =
{
#ifdef CONFIG_SSD1306_UNICODE_ENABLE
//  type|width|height|first char
    0x01, 0x0D, 0x0B, 0x20,
//  unicode(2B)|count
    0x20, 0x00, 0x60, // unicode record
#else
//  type|width|height|first char
    0x00, 0x0D, 0x0B, 0x20,
#endif
    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char ' ' (0x0020/32)
    0xBE, 0xBE, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '!' (0x0021/33)
    0x0E, 0x00, 0x0E, 0x0E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '"' (0x0022/34)
    0x00, 0x60, 0x78, 0xF8, 0x7E, 0xFA, 0x7E, 0x7E, 0x18, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '#' (0x0023/35)
    0x00, 0x00, 0xDC, 0xBE, 0xFF, 0xF6, 0xEC, 0xC4, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x07, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '$' (0x0024/36)
    0x00, 0x04, 0x0E, 0x8A, 0x4E, 0x20, 0xD8, 0xEC, 0xE2, 0xC0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, // char '%' (0x0025/37)
    0x00, 0x40, 0xFE, 0xBE, 0xBE, 0xEE, 0xE0, 0xF0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, // char '&' (0x0026/38)
    0x0E, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char ''' (0x0027/39)
    0x00, 0x70, 0xFE, 0x8E, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '(' (0x0028/40)
    0x00, 0x02, 0xDE, 0xFC, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char ')' (0x0029/41)
    0x00, 0x04, 0x0E, 0x0E, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '*' (0x002A/42)
    0x00, 0x30, 0x78, 0x78, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '+' (0x002B/43)
    0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char ',' (0x002C/44)
    0x60, 0x60, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '-' (0x002D/45)
    0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '.' (0x002E/46)
    0x00, 0x00, 0x80, 0xC0, 0xF0, 0x38, 0x1C, 0x06, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '/' (0x002F/47)
    0x00, 0x20, 0xF8, 0xFC, 0x8C, 0xFC, 0xF8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '0' (0x0030/48)
    0x00, 0x30, 0xB8, 0xFC, 0xFC, 0x80, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '1' (0x0031/49)
    0x00, 0x08, 0xD8, 0xCC, 0xEC, 0xFC, 0xB8, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '2' (0x0032/50)
    0x00, 0x10, 0x38, 0xD8, 0xD8, 0xF8, 0xB0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x04, 0x06, 0x06, 0x06, 0x07, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '3' (0x0033/51)
    0x00, 0x80, 0xE0, 0xF0, 0xF8, 0xFC, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x03, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '4' (0x0034/52)
    0x00, 0x80, 0xF8, 0xF8, 0xD8, 0xD8, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x07, 0x06, 0x06, 0x07, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '5' (0x0035/53)
    0x00, 0x30, 0xFC, 0xF6, 0x9E, 0xFE, 0xF4, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '6' (0x0036/54)
    0x00, 0x0C, 0x0C, 0xCC, 0xFC, 0x3C, 0x0C, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0x03, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '7' (0x0037/55)
    0x00, 0x40, 0xFC, 0xFE, 0xB6, 0xFE, 0xFC, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '8' (0x0038/56)
    0x00, 0x60, 0xF8, 0xD8, 0xD8, 0xF8, 0xF0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x06, 0x06, 0x07, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char '9' (0x0039/57)
    0x98, 0x98, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // char ':' (0x003A/58)
[... cut ...]
    /* Replace the line below with ssd1306_128x32_i2c_init() if you need to use 128x32 display */
    //ssd1306_128x64_i2c_init();
    ssd1351_128x128_spi_init(3, 4, 5);
    ssd1306_fillScreen(0x00);
    //ssd1306_setFreeFont(free_calibri11x12);
    ssd1306_setFreeFont(Raleway_Black13x11);
    //ssd1306_setSecondaryFont(free_calibri11x12_cyrillic);
    ssd1306_printFixed(0,  8, u8"This is Русский", STYLE_NORMAL);

    //ssd1306_setSecondaryFont(free_calibri11x12_latin);
    ssd1306_printFixed(0, 24, u8"This is Deutsch Länder", STYLE_NORMAL);
}

void loop()
{
}

sorriso93, 2018-09-13:
I just need a bigger font, not important which one...

lexus2k, 2018-09-13:
I found several issues:
1. The bug in the library, when secondary font is not set. I fixed it with latest commit.
2. The issue with font generator for old format font. It is fixed also.
3. There is bug in the sketch, you provided, since you generated old format font, but try to set the font via `ssd1306_setFreeFont()` (which can be used only for new fonts).

Please, pull latest fixes, regenerate the font (either old or new format) and use correct function to set the font.

lexus2k, 2018-10-02:
Can you confirm, that the code works for you now?

sorriso93, 2018-10-02:
Hi, right now I'm working with another tft display and I paused the oled ST1351 tests. As soon as I have spare time I'll let you know. Many thanks

lexus2k, 2018-10-14:
I closed the ticket. But feel free to reopen it, if you stil have any issues.

closed 2018-10-14 by lexus2k