lexus2k/ssd1306 #47: Redefining the pins for Atmega32u4 not working?

StackerDEV, 2018-08-27:
Please provide the following information:
## library version
ssd1306-1.7.6
## LCD display type
I2C SSD1306 128x64
## Steps to reproduce the issue
I tried chaning the pins in ssd1306_i2c_conf.h for my Atmega32u4 @ 16MHz. But it does not display anything when pins changed and code compiled again. Working fine on the Atmega328 @ 16MHz using pin A4 and A5 with I2C. I tried 18, 19, 7,8 and 0,1 (switches both once, just in case).
## Expected behavior
It should display text when uploading the test example.
## Actual behavior
Display stays black.

lexus2k, 2018-08-27:
Hello,

Well, for you platform (Atmega32u4) the library uses Wire module with standard i2c pins.
Please, read this thread https://forum.arduino.cc/index.php?topic=186965.0.
I don't think, that Wire library will work on other than standard pins for your board.
There is also software implementation in ssd1306 library, you can try it.

lexus2k, 2018-08-28:
Fortunately, I found Arduino Micro reference board, soldered it, and connected to ssd1306 i2c oled without any changes in ssd1306 library.
Just use correct i2c pins (2, 3) for your Arduino Micro board.

[image]

lexus2k, 2018-08-30:
@StackerDEV 

Did you have a chance to check the pins like on the picture above?

lexus2k, 2018-09-02:
I'm closing the issue since there is no any feedback.
And the tests say that ssd1306 display works on Arduino Pro Micro on 2/3 pins.

StackerDEV, 2018-09-05:
@lexus2k sorry fot the late reply. I needed pin 2 ad 3 for another component. I didn't try it because I think i haven't tried that, I really needed those pins already (this is why). Using a Arduino nano v3 with a ATmega328P now. Everything works like I intended it to work 2 seperate modules both using ic2. Thanks for testing and the feedback anyway. I use this lib and I like it. The only thing I still don't really understand is how to draw a canvas (filled box). I tried and almost got it the way I wanted but it was buggy because I was doing something wrong presumably. Also I found out, when I draw a line for example as indicator of a int value. When I update the width of the line it doesn't update, so I had to clear the entire screen, my workaround was to write char(0x20) in that line to "clear" it, lol.

lexus2k, 2018-09-05:
I reopened the issue to continue discussion. In order to workaround problem with pins on Atmega32u4, you can try to use software implementation of SSD1306 library.
As for flickering effect, you need to use canvas (but canvas will increase memory usage). There are examples in the library: double_buffering folder. And you can check NanoEngine (which use less memory, but a little more complicated: nano_engine/nano_engine)

StackerDEV, 2018-09-06:
@lexus2k awesome, thanks. I will try double buffering, it was kinda obvious actually now when you say it that I needed this for my project, hah.  Will order a new micro pro board (already ordered prototype boards 50x70mm), because I messed up mine because I only had 1 prototype board left and fully soldered the pro micro. When desoldering I found out that my new chinesium copper soldering wick (guutwick) was actually a complete waste of money and that I wasn't to get off the board off without the need of a hot air station that I don't have. So I had to sacrifice the board by cutting it off with pliers, lol. When I finish the project I will post my projectson GitHub, I do have other small projects that I want to finish kinda and china post isn't that fast so it could take a while for me to post and/or reply here.

lexus2k, 2018-09-11:
Did you figure out, how to use canvas objects?
Was it helpful for you?

closed 2018-09-17 by lexus2k

StackerDEV, 2018-09-19:
@lexus2k nope, not yet still waiting for my Micro Pro 32u4. But I do have all other components.