lexus2k/ssd1306 #48: Communication with other I2C devices

phKlip, 2018-09-01:
Hello
my main question is: How can a controller communicates with other I2C devices while using this library?

## library version
1.7.6

## LCD display type
I2C SSD1306 128x64

## Steps to reproduce the issue
I´m trying to communicate between an ATTiny 85 (master), a display (slave) and another device (slave, reader) via I2C. For the interaction with the another device (which is at the moment an Arduino for testing) I use the Wire library from Arduino. Therefore I have also changed the I2C pins in the ssd1306_i2c_conf file to the pins which are used by the Wire library.

## Expected behavior
The display works and the ATTiny sends data to the Arduino.

## Actual behavior
When I start the programm on the ATTiny, only the display works and the Arduino does not get any data.
(If I use the code without the display part, the Arduino will get the correct data).

many thanks

lexus2k, 2018-09-02:
Hello,

From your description I have several questions:
1. As far as I know, Wire library is not supported for Attiny85 chips? Am I wrong?
2. Which EVK do you use in your project? Digispark?
3. For Damellis Attiny85 board ssd1306 library supports only Software I2C implementation.
4. How do you initialize ssd1306 library in your project? Remember, that there are several ways to do that with ssd1306 library:
a) to use single function like `ssd1306_128x64_i2c_init()`, which selects some default i2c implementation for the project board (default not always means correct)
b) to use 2 separate functions like `ssd1306_platform_i2cInit()` (or any other supported i2c implementation) and `ssd1306_128x64_init()` (to initialize display itself).

Could you please send me `setup()` function content from your project?

phKlip, 2018-09-02:
Hello,
thanks for helping a programming beginner.

1. I´m using the ATTinyCore modul from Spence Konde. This modul have supported the Wire library since version 1.1.3, I think. 
Is that wrong, that I´m not using the Damellis ATTiny modul?

2. Sorry what does EVK mean? Evaluation kit? I had the display modul from AZdelivery (with for pins: Vcc, GND, SDA, SCL) and a single ATTiny 85 which is programming over the ISP of the Arduino.

3. See number one

4. Until now I have tried only the method a). But I also read the documentation for the function `void ssd1306_i2cInit(void)`:  "If you prefer to use your own Wire settings or avoid reinitializing of Wire library, please use ssd1306_i2cInit_Wire()", but these function does not exist.
Which file must be included to try it with the `ssd1306_platform_i2cInit()` function?

Here is my whole testing code for the ATTiny:
```
#include <Wire.h>
#include "ssd1306.h"

byte x = 0;

void setup() {
    Wire.begin();
    ssd1306_platform_i2cInit(2, 0, 0);
    ssd1306_128x64_init();
    ssd1306_fillScreen(0x00);
    ssd1306_setFixedFont(ssd1306xled_font6x8);
    
}

void loop() {
  ssd1306_printFixed (0,  8, "Hallo", STYLE_NORMAL);
  delay(500);
  ssd1306_printFixed (0,  8, "     ", STYLE_NORMAL);
  
  Wire.beginTransmission(8); 
  Wire.write(x);              
  Wire.endTransmission();    
  x++;
  delay(500);
}
```

Today I´m discovering, when I do not change the SDA and SCL pin in the ssd1306_i2c_conf file and connected all the devices as on the picture, the code is running, but on two different I2C lines. 
**(my target is to use only own I2C line)**

[image]

The next picture shows the diagramm of previous attempts, where only the display works and the arduino gets no data. (Here was the pin layout changed in the ssd1306_i2c_conf file and tested with the code from number four, too)

[image]

lexus2k, 2018-09-03:
It seems, I know, what happens. ATTinyCore module from Spence Konde supports Wire library for Attiny85. The ssd1306 library assumes that Wire library is not available for Attiny85 and uses software implementation to control display: software implementation conflicts with Wire library.
I installed ATTinyCore module to my Arduino IDE and made some changes to compilation to enable Wire for the board configuration you're using. Could you please check the changes by manually updating the library (let me know if you need help with this), I committed update to master branch

phKlip, 2018-09-04:
Now it works perfectly and you have solved the problem.
(The display and the arduino are both connected to pin 5 and 7)

Thank you very much.

closed 2018-09-04 by phKlip