lexus2k/ssd1306 #49: SH1106 fix

ivm164, 2018-09-15:
Please provide the following information:
## library version
1.7.6
## LCD display type
SH1106
## Steps to reproduce the issue
Just print any text
## Expected behavior
Text is fully visible
## Actual behavior
First two pixel columns are missing

To fix this you should change the lines number 73 and 74 in oled_sh1106.c file like this:
    ssd1306_intf.send(((x+2)>>4) | SSD1306_SETHIGHCOLUMN);
    ssd1306_intf.send(((x+2) & 0x0f) | SSD1306_SETLOWCOLUMN);

lexus2k, 2018-09-16:
@ivm164 

Thank you for reporting the issue. Is your display (visible area) 128 pixels and not 132 as SH1106 controller supports?
I integrated fix 1f7123e. Coud you please check it?

ivm164, 2018-09-17:
Hi Aleksei,

The fix works perfect for me. And you're right that the visible area seems
to be 128p. BTW I've used 10 sh1106 lcds from 3 different vendros and all
had the same hw config.

WBR

lexus2k, 2018-09-17:
Thank you very much for cooperation.

closed 2018-09-19 by lexus2k