lexus2k/ssd1306 #50: Bigger font ssd1351, Arduino M0 Pro

FrogManBird, 2018-09-21:
Please provide the following information:
## library version   9d74c90
## LCD display type  1.5 inch 128x128 ssd1351
## Steps to reproduce the issue  N/A
## Expected behavior  N/A
## Actual behavior  N/A

The good news is that I have this up and running on an Arduino M0 Pro (which I'm sad that they've retired because it's great).  The bad news is that I can't figure out how to get bigger font than the 6X8 using:
ssd1306_setFixedFont(ssd1306xled_font6x8);
ssd1306_printFixed8(0,  16, "Test String", STYLE_NORMAL);

I see routines like  ssd1306_printFixedN in the code, but I can't seem to get them to work on this system.  It's probably operator-error on my part.  Can you show me the light on this?

lexus2k, 2018-09-22:
Hello,

Could you please send me compilation command for the files of ssd1306 library (you can get it if enable verbose mode in Arduino IDE)?
Does `direct_draw/draw_text` work well for you? If not, can you take a picture of lcd content?

CrowMagnumMan, 2018-09-22:
Hi,
FrogManBird is my old account.  Anyway, I got it working now.  You guys did a really outstanding job on this.

direct_draw/draw_text did indeed work fine.  Then I realized a few things, mainly that this statement was creating a mess:
ssd1306_setMode( LCD_MODE_NORMAL );  //this caused screen mess problem, using ssd1351 no better

Long story short, here's a version of your draw_text program that I tweaked to work on my system, this one showing how to use colors as well:

#include "ssd1306.h"
void setup()
{
    //works with Arduino M0 Pro running a 1.5 inch 128x128 ssd1351 display
    //MOSI and SCK connected to ICSP header. Other 3 SPI signals connected to digital pins below 
    ssd1351_128x128_spi_init(3, 4, 5);  //RST, CS, DC
    ssd1306_setFixedFont(ssd1306xled_font6x8);
    ssd1306_clearScreen();
    ssd1306_printFixed (0,  8, "Line 1. Normal text", STYLE_NORMAL);
    ssd1306_setColor(RGB_COLOR16(255,0,0));  //must use COLOR16 witb ssd1351
    ssd1306_printFixed (0, 16, "Line 2. Bold text", STYLE_BOLD);
    ssd1306_setColor(RGB_COLOR16(255,255,0));
    ssd1306_printFixed (0, 24, "Line 3. Italic text", STYLE_ITALIC);
    ssd1306_setColor(RGB_COLOR16(255,0,249));
    ssd1306_printFixedN (0, 40, "DoubleSize", STYLE_NORMAL, FONT_SIZE_2X);
    ssd1306_setColor(RGB_COLOR16(252,147,3));
    ssd1306_printFixedN (0, 68, "4", STYLE_NORMAL, FONT_SIZE_4X);
    ssd1306_printFixedN (0, 96, "X", STYLE_NORMAL, FONT_SIZE_4X);
    ssd1306_setColor(RGB_COLOR16(16,239,33));
    ssd1306_printFixedN (30, 72, "8X", STYLE_NORMAL, FONT_SIZE_8X);    
}

void loop()
{
}

lexus2k, 2018-09-23:
Thank you.

ssd1306 library has 2 sets of functions: with `8` suffix (like ssd1306_printFixed8()) and without `8` suffix (like ssd1306_printFixed()). Functions without the suffix work in ssd1306 compatible mode, that is: GDRAM addressing mechanism is the same as for ssd1306 controller. Such addressing has some limitations.
Function with the suffix `8` work in native mode for selected display. By default, for any supported display the ssd1306 library works in ssd1306 compatible mode.

Unfortunately, `ssd1306_printFixedN8()` is not yet implemented. But you can switch between modes using `ssd1306_setMode()` as you already know.

closed 2018-09-27 by lexus2k