lexus2k/ssd1306 #51: console.print() issues

gcormier, 2018-09-29:
## library version
1.7.8
## LCD display type
i2c ssd1306

I can't figure out the console.

`  console.clear();
  console.print("Temp : ");
  console.println(12);
  console.print("Humidity : ");
  console.println(34);`
This outputs one line at the top of the screen, "Humidity : 34"

`  console.clear();
  console.print("Temp : ");
  console.println(12);
  //console.print("");
  console.print("Humidity : \n");
  //console.println(currentHumidity);
  console.println(34);`

This outputs "Humidity : (newline)34" and it scrolls down the screen, looping to the top after a hitting the bottom.

lexus2k, 2018-09-30:
Hello,

please, provide full sketch code. Maybe the root cause is the font being used. I tested your code and in all cases, I've got exactly what is expected.
[image]

gcormier, 2018-09-30:
```
void setup()
{
  Serial.begin(9600);
  Serial.println(F("Starting..."));
  pinMode(PIN, INPUT);
#ifdef DEBUG
  pinMode(DEBUGPIN, OUTPUT);
  digitalWrite(DEBUGPIN, LOW);
#endif

  ssd1306_128x64_i2c_init();
  ssd1306_clearScreen(); 
  ssd1306_setFixedFont(ssd1306xled_font6x8);
  console.begin();
 
  while (true)
  {
    updateDisplay();
    delay(250);
  }
}

void updateDisplay()
{
  console.clear();
  
  console.print("Temp : ");
  console.println(12);
  console.print("Humidity : \n");
  console.println(34);
  
}
```

This prints 

> Humidity : 
> 34

Any it scrolls downwards looping around at the top.

gcormier, 2018-10-01:
Great, thanks!

closed 2018-10-01 by gcormier