lexus2k/ssd1306 #52: ESP32 component integration

HelixX23, 2018-10-18:
Hi,
i am currently trying to integrate your SSD1306 library into an ESP-IDF C project.
I am using an I2C SSD1306 128x64px OLED Display on an ESP32 dev board.

I copied the "ssd1306" folder which includes the "src" folder into the projects "components" folder.
I can reference the library but as i compile it, the reference of the functions defined in ssd1306_generic.c cannot be found.

I included the following files 
#include "ssd1306.h"
#include "ssd1306_fonts.h"

Inside that "ssd1306_fonts.h" header file the missing functions are defined.

The function "ssd1306_128x64_i2c_init()" can be found
the functions
ssd1306_setFixedFont()
ssd1306_fillScreen()
ssd1306_setCursor()
ssd1306_write()
cannot...

is there something i missed?
best regards

HelixX23, 2018-10-18:
as i tried to run 
`./build_and_run.sh -p esp32 demos/ssd1306_demo`
(in /PATH/TO/MY/PROJECTS/COMPONENTS/FOLDER/ssd1306/tools/)
i get many lines reporting about missing references issue:
`/PATH/TO/MY/PROJECTS/COMPONENTS/FOLDER/ssd1306/bld/esp32/build/sketch/libsketch.a(ssd1306_demo.o):(.literal._ZL10bitmapDemov+0x4): undefined reference to 'ssd1306_drawBitmap'`

lexus2k, 2018-10-18:
Hello.

I'm sorry for the bug. `./src` path is missed in library component.mk file.
Please, see the patch a7420e3 or clone the latest sources.
Thank you for reporting the issue.

HelixX23, 2018-10-18:
Thanks for the quick reply.
It compiles correctly now :+1:

HelixX23, 2018-10-18:
Very nice Demo! btw

closed 2018-10-19 by HelixX23