lexus2k/ssd1306 #54: which ssd1306 library on the Arduino IDE library manager to use?

dportabella, 2018-10-22:
I'd like to use the ssd1306 arduino library for my "ESP8266 0.96''OLED ESP-12F WIFI Development Board":
https://www.ebay.es/itm/ESP8266-0-96OLED-ESP-12F-WIFI-Development-Board-CP2102-Micro-USB-18650-Battery/113002221260

Your README.md file says:

> Setting up
>   Setting up for Arduino: (variant 2)
>     Install ssd1306 library via Arduino IDE library manager
> 

there are many ssd1306 libraries on the Arduino IDE library manager.
for instance: Acrobotic SSD1306, Adafruit SSD1306, ESP8266 and ESP32 Oled Driver for SSD1306 display...

Which one of those libraries are compatible?

lexus2k, 2018-10-22:
Hello,
This github account is dedicated to ssd1306 by Alexey Dynda. See picture below:
[image]

lexus2k, 2018-10-22:
This account is dedicated to ssd1306 by Alexey Dynda. See picture below:
[image]

closed 2018-10-24 by lexus2k

dportabella, 2018-10-24:
maybe you can add this to the README.md file...

lexus2k, 2018-10-27:
Added.