lexus2k/ssd1306 #56: custom_font_unicode display problem

choujayyl, 2018-11-05:
Please provide the following information:
## library version
     master
## LCD display type
     ssd1306  128x32 i2c
## Steps to reproduce the issue
      run demo custom_font_unicode
## Expected behavior
     `    // First variant of usage: using unicodes with standard ascii font
    ssd1306_setFixedFont( ssd1306xled_font6x8 );
    ssd1306_setSecondaryFont( g_customUnicodeFont_6x8 );
    ssd1306_printFixed (0,  8,  "Ascii + Ää", STYLE_NORMAL );

    // Second variant of usage: without standard ascii font
    ssd1306_setFixedFont( g_customUnicodeFont_6x8 );
    ssd1306_printFixed (0,  16,  "ÄäÄäÄä", STYLE_NORMAL );`
## Actual behavior
     
[image]

lexus2k, 2018-11-05:
Hello,

I'm sorry for that. There 2 two bugs in the code. 
Thank you for reporting, please, check f5a364c changes

lexus2k, 2018-11-06:
Could you please, confirm, that the changes work for you?

closed 2018-11-07 by lexus2k