lexus2k/ssd1306 #63: 128x32 View Aspect

phodgers, 2019-01-04:
## library version
1.7.13 (unofficial)

## LCD display type
128x32 SPI

## Steps to reproduce the issue
When I run the demo ssd1306_demo from examples it displays text and images vertically compressed in the 128x32 display. I would think it should show in normal aspect but anything outside display size would be clipped and not shown?

I think to change the reference ssd1306_128x64_spi_init(n,n,n) to ssd1306_128x32_spi_init(n,n,n) in setup() section of example code, to correctly define my display, but I find there is no such call. Is there somewhere else I should be changing a setting or parameter to define the size?

Thanks

[image]
[image]

lexus2k, 2019-01-04:
Hello,

The case happens, because initialization steps for ssd1306 controller are slightly different for 128x64 and 128x32 displays (multiplex command for those displays is different). The  `ssd1306_128x64_spi_init()` is defined like this:

```.cpp
void   ssd1306_128x64_spi_init(int8_t rstPin, int8_t cesPin, int8_t dcPin)
{
    if (rstPin >=0)
    {
        ssd1306_resetController( rstPin, 10 );
    }
    ssd1306_spiInit(cesPin, dcPin);
    ssd1306_128x64_init();
}
```

I didn't implement spi initialization for 128x32 oled, but you can use the code below to make ssd1306 128x32 working in your case:
```.cpp
    if (rstPin >=0)
    {
        ssd1306_resetController( rstPin, 10 );
    }
    ssd1306_spiInit(cesPin, dcPin);
    ssd1306_128x32_init();
```

lexus2k, 2019-01-04:
Check commit f040280

phodgers, 2019-01-04:
That works great now thanks.
[image]

closed 2019-01-04 by phodgers