lexus2k/ssd1306 #65: Using ssd1306_clearScreen() is slow

chutsu, 2019-01-31:
Please provide the following information:
## library version
Master branch - commit version 54c9888

## LCD display type
Adafruit SSD1351

## Steps to reproduce the issue
```
ssd1351_128x128_spi_init(25, 0, 24);
ssd1351_setMode(LCD_MODE_NORMAL);
ssd1306_setFixedFont(ssd1306xled_font6x8);
ssd1306_clearScreen();
```

## Expected behavior
Was expecting the `ssd1306_clearScreen();` to clear the screen instantly.

## Actual behavior
The display is cleared line by line, top to bottom. See video [here](https://www.youtube.com/watch?v=uQS7h_E0TDM)

lexus2k, 2019-01-31:
Hello,

Did you try Adafruit SSD1351 library? Is it faster?
Are you using Arduino Nano, or something else (ESP32/Raspberry)?

To fill the screen content, it is required to send at least 32KiB of data. I'm telling "at least", because except payload, library needs to send commands.  That's about 300kbit of data to clear the screen. Even at max available speed Arduino Nano gives only ~ 5-6 frames over SPI. 
So, reducing SPI speed (using software implementation instead of hardware) greatly slows down display operations.

chutsu, 2019-01-31:
I'm actually driving the display via a Raspberry Pi Zero W. I've tried using [Luma OLED](https://github.com/rm-hull/luma.oled) and have been able to instantly draw and clear the screen without any problems. I needed a C++ version for the display driver thats why I chose your library.

lexus2k, 2019-02-01:
Ok, I see.

I need to debug on my setup (raspberry pi 2+). According to this forum https://www.raspberrypi.org/forums/viewtopic.php?t=177965 ioctl in ssd1306 library sets write speed correctly.
Did you try spidev_test, described here: https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting. What are the results?

lexus2k, 2019-02-01:
I reproduced described behavior  on my raspberry, now debugging. As soon as I have solution, I will let you know

chutsu, 2019-02-02:
Thanks! Can confirm the fix does indeed fix the slow update.

closed 2019-02-02 by chutsu