lexus2k/ssd1306 #68: Support for use as IDF component

montaguk, 2019-03-06:
To start, I must say that this is an awesome project!  I was able to get my OLED display up and running on an ESP32 devboard using IDF with minimal effort.  Thank you for your awesome contributions!

That said, I couldn't quite figure out how to use the library as an IDF component as-is, and ultimately had to fork the repo and make some minor modifications to the ESP platform driver for the SSD1306 to get things running.  It seems that the crux of the my issue is that the setup arguments are not being passed correctly through ssd1306_i2cInitEx to the platform driver.

It looks like the ESP version of the platform init function ([ssd1306_platform_i2cInit](https://github.com/lexus2k/ssd1306/blob/dc6155631e36d023ba07c9daa2000f1bfaaf9178/src/ssd1306_hal/esp/platform.c#L118) is being called with incorrect parameters from the generic I2C init routine, [ssd1306_i2cInitEx](https://github.com/lexus2k/ssd1306/blob/master/src/intf/i2c/ssd1306_i2c.c#L31).

From what I can tell, there are a number of issues here:
1. `ssd1306_platform_i2cInit` currently expects to be called with a busId, i2c address, and some "args", but is actually being passed scl, sda, and the i2c address (sa).
2. `ssd1306_platform_i2cInit` doesn't even try to allow the user to specify the scl/sda pin IDs, but instead always uses 21/22.

My changes were relatively minor to get it up and running, but it seems like there may be a larger issue at play here. While digging around I noticed that a couple of the platform drivers may exhibit similar issues.

So ultimately my question is: am I missing something?  I see in the README that ESP is one of the tested platforms, but from what I've found the current implementation would only work on one design that is defaulted in the platform implementation.

Would you be willing to accepting a pull request with some minor clean up of the ESP platform init function?  I'm not sure what exactly a robust solution would look like, but I'd be open to contributing some time to help sort it out.

Thanks again for the awesome library!

lexus2k, 2019-03-06:
Hello,

Thank you for the kind words. I agree that `ssd1306_platform_i2cInit()` has the issue with assigning user-specified pin numbers for esp32. `void ssd1306_platform_i2cInit(int8_t busId, uint8_t addr, int8_t arg);` is declared globally in src/ssd1306_hal/io.h and is common for all platforms, supported by the library
Possibly, the way out here is to declare new function for ESP32 with busId, scl pin, sda pin and address.
Any suggestions and pull requests are welcome.

montaguk, 2019-03-07:
It looks like the Arduino implementation uses a different signature for [ssd1306_platform_i2cInit](https://github.com/lexus2k/ssd1306/blob/8a417fea4bb3f46ca01109c84257e386a3fa27fd/src/ssd1306_hal/arduino/platform.cpp#L104) that takes the scl, sda, and sa arguments.  The remaining platforms seem to be using the common one though.

I also did a quick search through the various implementations of this routine and didn't see any that are actually using the args parameter.  It seems like a really simple fix would be to expand that to a uint32_t and use a bitwise operation or union to pack the scl, and sda params.  This might even allow the Arduino implementation to move to the common prototype.

If that approach seems reasonable to you, I could probably find some time to throw something together in the next couple days or so.

lexus2k, 2019-03-07:
All platforms use different signature in implementation, for example, Linux uses busId and deviceId. But busId and deviceId are not required for Arduino platform, and sda/scl are not required for Linux platform.
The main idea of platform API is to have single cross-platform declaration in [ssd1306_hal/io.h](https://github.com/lexus2k/ssd1306/blob/master/src/ssd1306_hal/io.h) and to use it in ssd1306_i2cInitEx. 
And also, the library allow to have separate signature and implementation, if it is required to implement some specific behavior.

Arduino implementation uses arg parameter as sda in `void ssd1306_platform_i2cInit(int8_t scl, uint8_t sa, int8_t sda)`, but as for esp32, I agree that arg parameter can be used as argument to control pins.

lexus2k, 2019-03-25:
Hello, I provided some review comments. Could you please, check them?

montaguk, 2019-03-25:
I'm not seeing any review comments on the most recent PR (#74).  Should I be looking somewhere else?

closed 2019-04-08 by lexus2k

chandanchan0086, 2021-05-07:
can any one please send me example code to implement SSD1306 with esp-32 using esp-idf