lexus2k/ssd1306 #69: XBitMap (xbm) for ssd1306_drawBitmap produces garbage on OLED

zerkalica, 2019-03-07:
How to prepare image for ssd1306_drawBitmap?

I try to convert image to indexed 1bit color xbm (via gimp) and place array into [draw_bitmap](https://github.com/lexus2k/ssd1306/blob/master/examples/direct_draw/draw_bitmap/draw_bitmap.ino) example. I keep example data types (uint8_t Owl [] PROGMEM). ssd1306_drawBitmap produce garbage on my OLED.

Something like u8g2.drawXBMP?

bl.xbm

```
#define bl_width 107
#define bl_height 64
static unsigned char bl_bits[] = {
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xfa, 0xff, 0x03, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0xff, 0xff, 0x0f, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xe0, 0xfe, 0x1c,
   0x18, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xc0,
   0xff, 0x1f, 0x3e, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0xc0, 0x1f, 0x00, 0x7e, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0xc0, 0x01, 0x00, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x3c, 0x00, 0x00, 0x00, 0x60, 0x00, 0x00, 0xe0, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x10, 0x66, 0x00, 0x00, 0x00, 0x20, 0x00, 0x0b, 0xe0, 0x03,
   0x00, 0x00, 0x00, 0x00, 0x08, 0xf3, 0x00, 0x00, 0x00, 0x30, 0xe0, 0x0d,
   0x80, 0x03, 0x00, 0x00, 0x00, 0x00, 0xcc, 0x99, 0x03, 0x00, 0x00, 0x18,
   0x18, 0x0f, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x66, 0x08, 0x07, 0x00,
   0x00, 0x18, 0xde, 0x07, 0x00, 0x07, 0x00, 0x00, 0x00, 0x00, 0x26, 0xc3,
   0x06, 0x00, 0x00, 0x18, 0x62, 0x8b, 0x01, 0x1e, 0x00, 0x00, 0x00, 0x00,
   0x32, 0xc1, 0x0c, 0x00, 0x00, 0x18, 0x20, 0xdd, 0x00, 0x1e, 0x00, 0x00,
   0x00, 0x00, 0x33, 0x0d, 0x1c, 0x00, 0x00, 0x18, 0xe3, 0x1f, 0x00, 0x7c,
   0x00, 0x00, 0x00, 0x00, 0x31, 0x6c, 0x36, 0x00, 0x00, 0x18, 0xfe, 0x0f,
   0x00, 0x7c, 0x00, 0x00, 0x00, 0x00, 0x10, 0x60, 0x30, 0x00, 0x00, 0x48,
   0xe0, 0x06, 0x3c, 0x7c, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x20, 0x00,
   0x00, 0xc8, 0x01, 0x01, 0x3e, 0x7c, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00,
   0x38, 0x00, 0x00, 0xc8, 0x03, 0x00, 0x27, 0xb8, 0x01, 0x00, 0x00, 0x00,
   0x10, 0x00, 0x38, 0x00, 0x00, 0x08, 0xc7, 0x8a, 0x01, 0xfc, 0x01, 0x00,
   0x00, 0x00, 0x10, 0x00, 0x18, 0x00, 0x00, 0xc8, 0xef, 0xcb, 0x00, 0xf8,
   0x03, 0x00, 0x00, 0x00, 0x30, 0x00, 0x18, 0x00, 0x00, 0xc8, 0xff, 0xff,
   0xff, 0x71, 0x06, 0x00, 0x00, 0x00, 0x60, 0x00, 0x0c, 0x00, 0x00, 0xcc,
   0x7b, 0xff, 0x19, 0xb0, 0x05, 0x00, 0x00, 0x00, 0xc0, 0x00, 0x06, 0x00,
   0x00, 0xc4, 0x8f, 0xc1, 0x2d, 0xf0, 0x04, 0x00, 0x00, 0x00, 0xe0, 0x01,
   0x03, 0x00, 0x00, 0xa6, 0xe7, 0xc0, 0x27, 0x60, 0x04, 0x00, 0x00, 0x00,
   0x60, 0x00, 0x02, 0x00, 0x00, 0xe6, 0x33, 0xe0, 0x3f, 0x60, 0x04, 0x00,
   0x00, 0x00, 0xb0, 0x01, 0x02, 0x00, 0x00, 0xc6, 0x11, 0x30, 0x0e, 0xc0,
   0x06, 0x00, 0x00, 0x00, 0x74, 0x03, 0x03, 0x00, 0x00, 0x06, 0x18, 0x30,
   0x00, 0xc0, 0x06, 0x00, 0x00, 0x00, 0xdc, 0x02, 0x03, 0x00, 0x00, 0x06,
   0x0f, 0x3c, 0x00, 0x40, 0x04, 0x00, 0x00, 0x00, 0x78, 0x00, 0x03, 0x00,
   0x00, 0x06, 0x03, 0x38, 0x00, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x48, 0x0e,
   0x03, 0x00, 0x00, 0x06, 0x03, 0x76, 0x00, 0x80, 0x0f, 0x00, 0x00, 0x00,
   0x9e, 0x0c, 0x02, 0x00, 0x00, 0x02, 0x7e, 0xbf, 0x00, 0x00, 0x03, 0x00,
   0x00, 0x00, 0xfc, 0x01, 0x02, 0x00, 0x00, 0x03, 0xfc, 0x9d, 0x01, 0x00,
   0x06, 0x00, 0x00, 0x00, 0xf6, 0x01, 0x06, 0x00, 0x00, 0x01, 0x90, 0x00,
   0x03, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x36, 0x01, 0x06, 0x00, 0x80, 0x01,
   0x1c, 0x00, 0x06, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x1f, 0x02, 0x04, 0x00,
   0x80, 0x01, 0x04, 0x3c, 0x04, 0x00, 0x18, 0x00, 0x00, 0x00, 0x6b, 0x83,
   0x0d, 0x00, 0x80, 0x01, 0xf0, 0xff, 0x0c, 0x00, 0x30, 0x1f, 0x00, 0x00,
   0x72, 0x80, 0x09, 0x00, 0x80, 0x01, 0xf0, 0xbf, 0x1b, 0x00, 0xf0, 0x3f,
   0x00, 0x00, 0x73, 0x00, 0x18, 0x00, 0x00, 0x01, 0xe0, 0x03, 0x1a, 0x00,
   0xe0, 0xe0, 0x00, 0x00, 0x83, 0x03, 0x10, 0x00, 0xfe, 0x03, 0x80, 0xf0,
   0x1f, 0x00, 0xe0, 0xc0, 0x03, 0x00, 0x82, 0x00, 0x30, 0x80, 0xff, 0x02,
   0x80, 0x1f, 0x16, 0x00, 0x20, 0x40, 0x1f, 0x00, 0x02, 0x00, 0x38, 0xc0,
   0x00, 0x06, 0xc0, 0xef, 0x01, 0x80, 0x21, 0xc0, 0xfc, 0x00, 0xe3, 0x00,
   0x6c, 0x60, 0x00, 0x0c, 0xe0, 0x37, 0x00, 0x00, 0x23, 0xc0, 0xc0, 0x01,
   0x4e, 0x00, 0xe4, 0x30, 0x00, 0x18, 0xf0, 0x1c, 0x00, 0x00, 0x23, 0xc7,
   0x00, 0x07, 0x92, 0x01, 0xb2, 0x18, 0x00, 0x30, 0x30, 0x0f, 0x10, 0x00,
   0x33, 0xcd, 0x00, 0x04, 0xc6, 0x80, 0xbf, 0x0e, 0x00, 0x60, 0xe0, 0x01,
   0x10, 0x00, 0xb1, 0xd1, 0x00, 0x00, 0x06, 0xc0, 0x9e, 0x07, 0x00, 0xc0,
   0xc1, 0x06, 0x1c, 0x80, 0x91, 0xf0, 0x00, 0x00, 0xd2, 0x70, 0xcb, 0x01,
   0x00, 0x40, 0x07, 0xfe, 0x07, 0xc0, 0xd8, 0xe0, 0x00, 0x00, 0x1b, 0xdc,
   0xed, 0x1b, 0x00, 0x40, 0x3c, 0x70, 0x98, 0x7f, 0x4c, 0x40, 0x00, 0x00,
   0xff, 0x47, 0x36, 0x1a, 0x00, 0xc0, 0xf0, 0xff, 0xff, 0x1f, 0x6c, 0x00,
   0x00, 0x00, 0xff, 0x86, 0x19, 0x1e, 0x00, 0x80, 0xb1, 0xff, 0x1f, 0x38,
   0x36, 0x00, 0x00, 0x00, 0xc1, 0xc3, 0x0c, 0x1c, 0x00, 0x00, 0x7b, 0x81,
   0x07, 0x3c, 0x1f, 0x00, 0x00, 0x00, 0xff, 0x60, 0x04, 0x1c, 0x00, 0x00,
   0xca, 0x83, 0x18, 0x8c, 0x07, 0x00, 0x04, 0x00, 0x03, 0x3c, 0x06, 0x1c,
   0x00, 0x00, 0x0e, 0xa7, 0x19, 0xdc, 0x03, 0x00, 0x06, 0x00, 0x01, 0x0f,
   0x03, 0x1c, 0x00, 0x00, 0x0e, 0x66, 0x3d, 0xfc, 0x01, 0x00, 0x02, 0x00,
   0xf1, 0xe1, 0x01, 0x0c, 0x00, 0x00, 0x04, 0x0c, 0x7a, 0xf3, 0x00, 0x00,
   0x02, 0x00, 0x70, 0x78, 0x00, 0x0c, 0x00, 0x00, 0x00, 0x38, 0xfc, 0x79,
   0x00, 0x60, 0x03, 0x00, 0x01, 0x0c, 0x00, 0x0c, 0x00, 0x00, 0x00, 0xe0,
   0xee, 0x3f, 0x00, 0x60, 0x01, 0x00, 0xff, 0x07, 0x00, 0x0c, 0x80, 0x01,
   0x00, 0x80, 0xff, 0x1f, 0x00, 0xe0, 0x01, 0x00, 0xfe, 0x01, 0x00, 0x0e,
   0x00, 0x01, 0x00, 0x00, 0xfc, 0x05, 0x00, 0xe0, 0x01, 0x00, 0x04, 0x00,
   0x00, 0x0e, 0x00, 0x01, 0x00, 0x10, 0xd8, 0x07, 0x00, 0xe0, 0x00, 0x00,
   0x0c, 0x00, 0x00, 0x06, 0x00, 0x01, 0x01, 0x10, 0xf0, 0x01, 0x00, 0xe0,
   0x00, 0x01, 0x18, 0x00, 0x00, 0x06, 0x80, 0x01, 0x01, 0x10, 0xe0, 0x01,
   0x00, 0xe0, 0x80, 0x03, 0x70, 0x00, 0x00, 0x00, 0x80, 0x01, 0x00, 0x00,
   0xc0, 0x01, 0x00, 0xc0, 0x00, 0x00, 0xe0, 0x00, 0x00, 0x00, 0x80, 0x03,
   0x00, 0x00, 0xc0, 0x03, 0x00, 0xc0, 0x00, 0x00 };
```

lexus2k, 2019-03-07:
Hello,

For now there is not function, supporting XBMP format. This can be added in future.
`ssd1306_drawBitmap()` uses native ssd1306 controller format, in which each byte represents eight vertical pixels. In XBMP format each byte represents eight horizontal pixels, that's why you see the garbage.
```.cpp
// The image consisting of two 0x13, 0x20 bytes will look like 
// ssd1306_Bitmap  XBMP
// X-------        XX--X---
// XX------        -X------
// --------        --------
// --------        --------
// X-------        --------
// --------        --------
// --------        --------
// --------        --------
```
Actually, the picture you will get depends on width and height values passed to the function

zerkalica, 2019-03-07:
Thanks.
Do you know simple offline converting tools for linux? 
I can't find ready to use offline tool for converting bitmap image to native format (like python script)

Only this online scripts worked:
http://www.majer.ch/lcd/adf_bitmap.php, http://www.rinkydinkelectronics.com/t_imageconverter_mono.php

lexus2k, 2019-03-08:
I added `ssd1306_drawXBitmap()` which supports monochrome XBMP images. Please, check.
To convert bmp picture to XBMP format I used script from https://github.com/robertgallup/bmp2hex
> ./bmp2hex.py -x -w 8 soba.bmp

zerkalica, 2019-03-08:
Yes, ssd1306_drawXBitmap works. But i use canvas.drawBitmap1, may be add same canvas.drawXBitmap1?

lexus2k, 2019-03-09:
Please, check latest commit. But I didn't test that function.

zerkalica, 2019-03-10:
I try to create running line, but canvas shows garbage, if x position + image width is greater than canvas size (canvas.drawBitmap1 works fine here).

Example: [xbm-bug.ino](https://gist.github.com/zerkalica/c45e264e08b85a5bc470ab4178230b42)

`canvas.drawXBitmap1(80, 0, 107, 64, s2 ); `

lexus2k, 2019-03-11:
Please, check the fix

zerkalica, 2019-03-11:
Thank you, my example works fine.

closed 2019-03-11 by zerkalica

lexus2k, 2019-03-11:
You're welcome