lexus2k/ssd1306 #70: drawLinesDemo Bug

AndreiCo14, 2019-03-16:
**Describe the bug**
See screenshots. Lines is broken.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload ssd1306_demo.ino
2. the result of the "draw lines" menu

**Expected behavior**
Draw lines.

**Screenshots**
[image]

**Please complete the following information:**
 - library version 1.7.16
 - LCD display ssd1306 0.96 I2C
 - OS windows
 - Platform wemos d1 mini
 - IDE Arduino 1.8.5

**Additional context**
-

lexus2k, 2019-03-17:
Hello,

Actually, this is not a bug, but expected behavior on these types of displays (based on ssd1306 controllers). This happens, because oled display doesn't allow to address single pixel. Each byte in ssd1306 controller represents 8 vertical pixels. So, for example, if it is required to draw horizontal line 4 pixels length at [0,2], library sends four bytes to ssd1306 (0x04,0x04,0x04,0x04). And then you need another horizontal line at [0,4], so the library needs to send another four bytes (0x10,0x10,0x10,0x10), and new bytes completely overwrite previous ones. So, you will see only one line, but not two.
Unfortunately there is no way to read old bytes from oled display of this type. So, the only way out for you here is to use double buffering, what is supported by ssd1306 library. 
Please, refer to examples.

lexus2k, 2019-03-17:
If you have any ideas how to solve your problem, let me know

AndreiCo14, 2019-03-17:
I understood. Thanks!

closed 2019-03-17 by AndreiCo14