lexus2k/ssd1306 #73: Console issue with 128x32 Oled

gbetous, 2019-03-20:
Hi !

You really wrote an amazing library :) I could see Arkanoid in a minute on ESP8266 + 128x32 OLED.

I'm simply trying to log simple messages in console mode and I see strange behaviors.

1°) Display is not the same if I `console.print` a string finishing with `\n`, compared to `console.println` without the trailing `\n`
2°) Using `console.println`, I only see one line (on the top of the screen) with the latest output. There is no "console" mode.
3°) Using `console.print` and trailing `\n`, I see these consecutive screens on my display :

```
Line 1
(empty)
(empty)
(empty)
```

```
Line 1
Line 2
(empty)
(empty)
```

```
Line 1
Line 2
Line 3
(empty)
```

```
(empty)
Line 2
Line 3
Line 4
```

```
Line 5
(empty)
Line 3
Line 4
```

```
Line 5
Line 6
(empty)
Line 4
```

And so on...
Not sure this is the expected behavior as a regular "console" always keeps latest line in the bottom.

Here is the sketch :

```
#include "ssd1306.h"
#include "ssd1306_console.h"

Ssd1306Console  console;

void setup(void){

 /* Replace the line below with the display initialization function, you want to use */
 ssd1306_128x32_i2c_init();
 ssd1306_clearScreen();
 console.setCursor(0,0);
 /* Set font to use with console */
 ssd1306_setFixedFont(ssd1306xled_font6x8);
 
}

char t[30];
int i=1;

void loop(void){
 sprintf(t, "(%03d) My message\n", i++);
 console.print(t);
 delay(1000);
}
```

Thank you very much,

Regards,

gUI

lexus2k, 2019-03-20:
Hello,

Thank you. The issue, you describe, is known issue. This happens, because Ssd1306Console class doesn't store screen content in memory buffer, but directly outputs all content to LCD display.  LCD displays, based on ssd1306 controller, do not allow to get content back. So, when log reaches the bottom line of the screen, the library cannot relocate old content up, and starts at the top line. That's what you see.

Actually, ssd1306 controller has scrolling hardware capabilities. I will read the datasheet, and let you know, if display content scrolling is possible via hardware features.

lexus2k, 2019-03-20:
It looks like it is possible to implement the feature you need. I need to do some tests with ssd1306 display.

gbetous, 2019-03-21:
Hi,

Thanks for your answer ! Don't hesitate ton involve me in some tests :+1:

lexus2k, 2019-03-21:
Please, check latest code on master branch.

Traace, 2019-03-21:
Thanks for the update, I updated my libs from master and also took a look into the example.

Still it doesn't work.
- Bottom lines get rendered before first lines
- /n spamming the whole display with its string before.
- 'println' doesn't work at all anymore.

Idk whats wrong :(

gbetous, 2019-03-21:
Thanks Aleksei for this update ! We are definitely moving forward :)

In opposite to Traace, I can see that we now are having a real console behavior, but still having some limitations :
- println still behaves like if there were a "clearScreen" (I can only see one line)
- using print with a '\n' ending string, the 10 first drawing are still a little bit weired
  - during the 8 first drawing, I can only see 3 lines of console (but scrolling is OK)
  - the 9th line is printed as the 4th line
  - there is a pause (like 3s)
  - after that drawing continues, on 4 lines, and scrolling still OK

Can you test on your own ? Do you want me to film ?

lexus2k, 2019-03-21:
Guillaume, thanks for the response.

I wrote simple test to check how linux console behaves with line endings, and I can say that lcd_console demo works exactly as linux script below on 2 oled displays: 128x64 i2c ssd1306 and 128x32 i2c ssd1306.
```.bash
#!/bin/sh

while true; do
   echo -n "Line\n"
   sleep 1
done
```

I also observe **freeze for 3 seconds**, even using SDL emulator, and yes, this is the issue with the library code. I didn't yet figure out, what's going on, still looking at the issue.

Maybe, some problems relate to specific hardware schematics: I saw different hardware implementation of oleds, based on the same ssd1306 controller. This could be a reason. I have 3 different monochrome oleds.

Moseeee, 2019-03-21:
> #!/bin/sh
> 
> while true; do
>    echo -n "Line\n"
>    sleep 1
> done

lexus2k, 2019-03-22:
The freeze is fixed

gbetous, 2019-03-23:
This is perfect, thanks a lot !

I guess that you can close this bug :+1:

lexus2k, 2019-03-23:
Thank you for the feedback.
Let me know if something else works wrong

closed 2019-03-23 by lexus2k