lexus2k/ssd1306 #76: Arduino Due: fatal error: avr/io.h: No such file or directory

rickyelqasem, 2019-04-02:
**Describe the bug**
Fails to compile on DUE board. I went back to basics and tried ssd1306 demo and it still fails

**To Reproduce**
Steps to reproduce the behavior:
1. Open the examples > ssd1306 > demos > ssd1306_demo
2. Compile

**Expected behavior**
Compile without failing.

**Screenshots**
NA

**Please complete the following information:**
 - library version: latest 1.7.18
 - LCD display type: OLED 128x32
 - OS : Windows
 - Platform Arduino Due
 - IDE if using some: Arduini.cc

**Additional context**
arduino\libraries\arduino_892937\src\intf\i2c\ssd1306_i2c_twi.c:31:20: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h>

lexus2k, 2019-04-03:
Thank you for reporting. I will double check

lexus2k, 2019-04-04:
Hello, can you check latest master branch: commit 085305a

commit 085305a9 (2019-04-04): Arduino Due compilation fix #76
closed 2019-04-08 by lexus2k