lexus2k/ssd1306 #78: composite_video.c:139:16: error: 'ssd1306_lcd_t' has no member named 'set_mode'

aldolo69, 2019-05-05:
**Describe the bug**

it seem last commit has broken something. my test progs and even demo do not compile anymore due to 
composite_video.c:139:16: error: 'ssd1306_lcd_t' has no member named 'set_mode'

**To Reproduce**
Steps to reproduce the behavior:

-compile, for example, the ssd1306_demo prog

**Please complete the following information:**
 - library version the very last
 - Platform  arduino uno 
 - arduino 1.8.5

lexus2k, 2019-05-05:
Thank you for reporting the issue.
Fixed per ae3339c commit.

closed 2019-05-06 by lexus2k

lucasromeiro, 2019-05-15:
How I can solve this? Is a balada problem for me

lexus2k, 2019-05-15:
@lucasromeiro 
What version do you use? Did you install ssd1306 from Arduino IDE manager or manually?

lucasromeiro, 2019-05-15:
> @lucasromeiro
> What version do you use? Did you install ssd1306 from Arduino IDE manager or manually?
> 
> Can you try 1.7_dev branch or master branch? They should be OK.

To solve I had to go back to version 17 17
I will wait for a new version to update with the fix.