lexus2k/ssd1306 #79: All fonts render garbage

amyspark, 2019-05-06:
**Describe the bug**
With current master, all fonts render garbage in my toy clock application. This bug was introduced in 67ffeb4a738e677fb93338ef1373ab4193400411 as per `git bisect` starting from `v1.7.12`; I don't know specifically what causes it.

**To Reproduce**
Upload the following sample to your board of choice (mine is an Arduino Uno, with a ST7735 LCD panel connected):

```c
#include <ssd1306.h>

void setup() {
  st7735_128x160_spi_init(8, 10, 9);
  ssd1306_fillScreen(0x00);
  ssd1306_setFixedFont(comic_sans_font24x32_123);
  ssd1306_printFixed(0, 0, "asd", STYLE_NORMAL);
}

void loop()
{

}

```

**Expected behavior**
I wanted to see "asd" in the screen. Instead, I get what is shown in the photo below.

**Screenshots**
[image]

**Please complete the following information:**
 - library version: ae3339ca4a94835c613f3f42ca342dec6a1a9bb9
 - LCD display type: st7735, 160x128, green tab
 - OS [e.g. linux, windows]: MacOS High Sierra 10.13.6
 - Platform [e.g. Atmega328p, esp32, etc.]: Arduino Uno (Atmega328p)
 - IDE if using some: Arduino 1.8.9, AVR toolchain v1.6.21

**Additional context**
Add any other context about the problem here.

lexus2k, 2019-05-07:
Thank you for reporting. I confirm, the issue exists

lexus2k, 2019-05-07:
Can you try the following code instead?

```.cpp
#include <ssd1306.h>

void setup() {
  st7735_128x160_spi_init(8, 10, 9);
  ssd1306_fillScreen(0x00);
  ssd1306_setFixedFont(comic_sans_font24x32_123);
  ssd1306_printFixed8(0, 0, "111", STYLE_NORMAL);
}

void loop()
{

}
```

amyspark, 2019-05-07:
I can confirm this works with `printFixed8`. What's the issue here?

lexus2k, 2019-05-09:
I reverted last 3 commits. So, now original example from your description should work as is.
The issue was related to the functionality, which I removed from the library. Thank you for reporting, I will think on how to do changes to the library more carefully.

closed 2019-05-10 by lexus2k