lexus2k/ssd1306 #81: st7735 fails to initialize on Arduino Mega 2560

amyspark, 2019-05-11:
**Describe the bug**
The library refuses to run on Arduino Mega 2560, it fails to initialize at all when using a ST7735-based TFT controller.

**To Reproduce**
Employ any of the provided examples. Point `rstPin` to pin 8, `cesPin` to pin 10, and `dcPin` to pin 9  in the relevant `st7735_128x160_spi_init` call, and wire SCK to 52 and SDA to 51.

**Expected behavior**
I expected anything (even garbage) to appear on the screen.

**Screenshots**
I don't have any, as the bug doesn't output any information.

**Please complete the following information:**
 - library version: 39f5f3f38e9009313cd8ec180ccf32cbc74f44e6
 - LCD display type: ST7735, 1.8'', 160x120 black tab
 - OS [e.g. linux, windows]: MacOS High Sierra 10.13.6/Windows 7
 - Platform [e.g. Atmega328p, esp32, etc.]: Atmega2560
 - IDE if using some: Arduino 1.8.9

**Additional context**
I've been able to reproduce this as far as 1.6.0 (first release to support st7735).
The Adafruit ST7735 examples also work with exactly the same wiring.

lexus2k, 2019-05-11:
Hello,

Did you try this display with other mcu and ssd1306 library? Could you please check it with Arduino nano, if you have one.
First, I need to understand if the issue relates to Atmega2560 only. I have some st7735 (128x160, don't know the manufacturer exactly), and it works with Arduino Nano and Esp32 Dev boards.
For now, I can only compare to Adafruit source code, and try to understand, what works differently.

lexus2k, 2019-05-11:
Please, check [1.7_dev](https://github.com/lexus2k/ssd1306/tree/1.7_dev) branch.
Let me know if something changed.

amyspark, 2019-05-12:
Nothing with 1.7_dev.

As for other MCUs-- my only other is an Arduino Uno (which is still AVR). I should have a STM32 blue pill in a few weeks, I could try then.

lexus2k, 2019-05-12:
1. What about other spi displays? Do you have monochrome ssd1306 spi oled? Are there any issues with it (in other words, is the issue st7735 only related)?
2. Could you please, comment out CONFIG_AVR_SPI_ENABLE in UserSettings.h and try again?
3. Please, check latest commit to 1.7_dev (don't forget to keep CONFIG_AVR_SPI_ENABLE disabled). This branch contains the same initialization sequence as Adafruit code with minimal difference.

If something of above works for you, I would like to ask you to help with debugging.

amyspark, 2019-05-12:
1. I don't have any other screen modules (yet, same as with the blue pill).
2. No output with any of the options.
3. No output with any of the options, although I did see in the pull that you changed the initialization code, but for the ILI9163. (Which, by the way, I should get along the blue pill.)
4. No output with any of the options.

lexus2k, 2019-05-12:
Here are my comments
2. Commenting out CONFIG_AVR_SPI_ENABLE disables internal SPI implementation in ssd1306 library, and makes it to use standard Arduino SPI implementation.
3. Actually il9163 and st7735 is the same controller, s_oled128x160_initData structure, which was modified, is used to initialize st7735 in 128x160 mode. When someone sales such display, you can see often both controllers mentioned "1.8 Inch 128X160 ILI9163/ST7735 TFT LCD Module With PCB Baseboard SPI Serial Port".
Thank you for the help. I will try to reproduce the problem on my setup.

lexus2k, 2019-05-13:
Please, check latest changes. At least on my st7735 128x160 led display, both Adafruit_ST7735 and ssd1306 (1.7_dev) works (with Arduino Nano).

lexus2k, 2019-05-15:
@amyspark Hello
Are there any updates?

amyspark, 2019-05-17:
Hey @lexus2k , I've been busy this week. I'll be able to test again in the weekend.

(As for testing in other mcu's -- I think they'll arrive by the end of the month.)

lexus2k, 2019-05-24:
@amyspark 

Thank you. Hope to see feedback soon.

amyspark, 2019-05-24:
Hey @lexus2k, it works now! I had to disable `CONFIG_AVR_SPI_ENABLE` and use Arduino's SPI implementation.

lexus2k, 2019-05-25:
Thank very much. I appreciate your help. 
I will prepare official fix soon

lexus2k, 2019-05-25:
It seems that I fixed built-in SPI issue for Atmega2560. But by default, I switch library to use Arduino's SPI implementation by default, since I don't have Atmega2560 by hand. You can check internal SPI implementation by commenting out CONFIG_PLATFORM_SPI_AVAILABLE.

amyspark, 2019-05-25:
@lexus2k, I can confirm your SPI implementation works now. Thanks for the fix!

lexus2k, 2019-05-26:
ok. Thank you.

closed 2019-05-26 by lexus2k