lexus2k/ssd1306 #86: Support for SSD1327

tbartel, 2019-07-04:
I'd like to use this library with an 1.5" SSD1327 grayscale OLED display. There are a handful of existing libraries with support, including U8G2 and https://github.com/bitbank2/ssd1327. Is there any chance support could be added in a future version?

lexus2k, 2019-07-05:
Hello @tbartel,
I can add support, and it would be great if you could help me to test it.
I don't have ssd1327 by hand.
Which version of ssd1327 do you have: i2c or spi?
Also, which library version would you like to use: 1.7.X or 2.X.X (it is easier for now to add support for 1.7.x)?

tbartel, 2019-07-06:
@lexus2k That's great! I have the SPI version of this display and would be happy to help test. Basing it off 1.7.x is fine with me.

lexus2k, 2019-07-06:
Please, check [1.7_dev](https://github.com/lexus2k/ssd1306/tree/1.7_dev) branch.
Can you run ssd1327_demo_compat example, and let me know, how it works

lexus2k, 2019-07-09:
Hello, did you have chance to check development branch?

tbartel, 2019-07-10:
I'm out of town until the weekend, so I won't have a chance to test for a few more days. I will update with the results as soon as I've had a chance to verify. Thank you for your quick work!

lexus2k, 2019-07-24:
Any updates on ssd1327 testing?

lexus2k, 2019-08-03:
Can you test ssd1327? If you don't need that support now, please let me know.

tbartel, 2019-08-03:
@lexus2k, sorry for being unresponsive. I was able to test ssd1327_demo_compat.ino today, but am seeing some sort of interlacing-type artifact:

https://imgur.com/MywhTPT

lexus2k, 2019-08-04:
@tbartel hi,

I really appreciate your help, you video is really helpful. I need to learning ssd1327 datasheet more in details. Will come back to you you with solution.

Thanks

lexus2k, 2019-08-04:
@tbartel I made the changes per fc9dabe commit. However, I'm trying to understand if your ssd1327 display use splitting odd/even of the COM signals or not. It depends on display manufacturer. Can you check my update?
Thank you

tbartel, 2019-08-04:
@lexus2k That seems to work! I'm able to see the owl image, heart sprite, etc. It does skip rendering every/other line, so that there's a 1px blank row between each 1px row drawn. This is most noticeable on the line drawing test. I will post a video a little bit later so you can better see what I mean.

tbartel, 2019-08-04:
@lexus2k Here's a video to better demonstrate my previous comment.

[image]

lexus2k, 2019-08-04:
Thank you very much.
I enabled com even/odd split feature. Can you test once more?

tbartel, 2019-08-05:
OK, I've given this new commit a test and it seems to be working properly. I'm attaching a short video so you can verify this is the expected output:

[image]

On a slightly unrelated note: given that the ssd1327 supports 4 levels of gray, is it possible to draw a sprite object containing multiple grays in a single sprite?

lexus2k, 2019-08-05:
@tbartel  Currently there is not such function (to draw a sprite object containing multiple grays in a single sprite), but I'm working on implementation.
As for the lines, there are dashed lines, because of specific of GDRAM layout, the way to workaround dashed lines is to use canvas.

lexus2k, 2019-08-05:
I added function `ssd1306_drawBitmap1_4`, the usage is available in ssd1327_demo_compat

tbartel, 2019-08-06:
Tested it out and confirmed it works. Thanks for the quick work! A couple of other questions:

- Is there a way to integrate this with the sprite structures?
- Is there a technical reason for sprite height being limited to 8 rows/a single byte? Given the resolution of the screen, it would be nice to build larger sprites. Perhaps the best solution to this is simply making a character composed of multiple sprites.

Finally, do you accept tips/donations anywhere? You've been incredibly quick and responsive in implementing these changes and I really appreciate it :-)

lexus2k, 2019-08-06:
Hi,

I fixed one more issue for `ssd1306_drawBitmap1_4`. However, as you noticed, I didn't implement yet NORMAL mode for ssd1327 display. NORMAL mode gives/will give more flexibility for multi-bit-displays like ssd1331, ssd1325, ssd1327 and others.

The limit in 8-pixel height for the sprites goes from sprites implementations as their functions are greatly optimized in speed. This is the only limitation for those functions. Sadly, I can't make  those functions applicable to 4-bit graphics.  But, there is another opportunity to make flickering-free graphics - to use NanoEngine. Let me know if you need 4-bit bitmaps support for canvas objects. And it is possible to implement some support for 4-bit graphics by adding new API.

 PS. I have accounts in [paypal](https://www.paypal.me/lexus2k), and [ko-fi](https://ko-fi.com/lexus2k), and I respect simple gratitude. :)

tbartel, 2019-08-06:
Thanks, I've tested it and it seems to work correctly. Canvas support for 4-bit bitmaps would be nice to have, but thanks to the existing changes you made I have enough to play with for now. Feel free to close this whenever you feel necessary.

lexus2k, 2019-08-08:
@tbartel Thank you for donation. I will use it to buy more displays :)
I released new version officially: 1.7.22. It can be downloaded via Arduino Library Manager or from github.

lexus2k, 2019-08-14:
Let me know if I can do anything else.
I'm working on 2.0 release for a while.

closed 2019-08-14 by lexus2k