lexus2k/ssd1306 #87: Garbage on screen

trollsid, 2019-07-14:
Test on ssd1351_demo

**Describe the bug**
Garbage on screen after run any command
**To Reproduce**
Steps to reproduce the behavior:
1. Compile and run
2. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**
 - library version 2.0_dev
 - LCD display type -ssd1351
 - OS : linux
 - Platform esp32
 - IDE : idf + vim

**Additional context**
Add any other context about the problem here.

lexus2k, 2019-07-15:
Hello, Dmitry.

Thanks for reporting. I don't have esp32 dev module by hand right now, but you could greatly help me, if you can check 1.7.x version and let me know if it works.

PS. Could you please post here display initialization code from your sketch? Thank you.

lexus2k, 2019-07-16:
Please,  check latest commit 5dd0328. It fixes spi issue with esp32 idf

lexus2k, 2019-07-17:
Fixed, feel free to reopen, if you have any issues

closed 2019-07-17 by lexus2k