lexus2k/ssd1306 #88: st7735_setRotation(3) 128x128 instead of 160x128

RoboDurden, 2019-07-25:
**Describe the bug**
With st7735_setRotation(3) 0,0 is somewhere at 24,0 and only pixels from 24,0 to 152x128 get printed to screen.

**To Reproduce**
```
    st7735_setRotation(1);
      ssd1306_drawRect8(1,1,159,127);
      delay(1000);
      ssd1306_printFixed8(0,  0, "123456", STYLE_NORMAL);
      delay(1000);
    st7735_setRotation(3);
      ssd1306_drawRect8(1,1,159,127);
      delay(1000);
      ssd1306_printFixed8(0,  0, "123456", STYLE_NORMAL);
      delay(1000);
```

**Screenshots**

Watch this 62 seconds video: https://youtu.be/CROUIsZCAJ0

**Please complete the following information:**
 - library version:  1.7.20 installed via Arduino IDE
 - LCD display type: cheap 1.8" ST7735
 - OS: windows 10
 - Platform: Atmega328p Arduino Pro Mini 5V 
 - IDE: Arduino 1.8.9

**Additional context**

With `#define st7735_setRotation il9163_setRotation` the bug should be here:
```
void il9163_setRotation(uint8_t rotation)
{
    uint8_t ram_mode;
    if ((rotation^s_rotation) & 0x01)
    {
        uint16_t t = ssd1306_lcd.width;
        ssd1306_lcd.width = ssd1306_lcd.height;
        ssd1306_lcd.height = t;
    }
    s_rotation = (rotation & 0x03) | (s_rotation & 0x04);
    ssd1306_intf.start();
    ssd1306_spiDataMode(0);
    ssd1306_intf.send(0x28);
    ssd1306_intf.send(0x36);
    ssd1306_spiDataMode(1);
    switch (s_rotation)
    {
    case 0:
        ram_mode = 0b00000000;
        break;
    case 1: // 90 degree CW
        ram_mode = 0b01000000;
        break;
    case 2: // 180 degree CW
        ram_mode = 0b11100000;
        break;
    case 3: // 270 degree CW
        ram_mode = 0b10000000;
        break;
    case 4:
        ram_mode = 0b00000000;
        break;
    case 5: // 90 degree CW
        ram_mode = 0b01100000;
        break;
    case 6: // 180 degree CW
        ram_mode = 0b11000000;
        break;
    default: // 270 degree CW
        ram_mode = 0b10100000;
        break;
    }
    ssd1306_intf.send( ram_mode | s_rgb_bit );
    ssd1306_spiDataMode(0);
    ssd1306_intf.send(0x29);
    ssd1306_intf.stop();
}
```

Please help :-)

lexus2k, 2019-07-25:
Hello.

Which mode do you use? COMPAT or NORMAL?

RoboDurden, 2019-07-25:
void setup()
{
    st7735_128x160_spi_init(3, 4, 5);
    ssd1306_setMode( LCD_MODE_NORMAL );
    ssd1306_clearScreen8( );
 }

I use the example https://github.com/lexus2k/ssd1306/blob/master/examples/demos/st7735_demo/st7735_demo.ino
Stripped to the textDemo() function.

00:30 here in Germany and already in bed. Will test other mode tomorrow.

lexus2k, 2019-07-26:
Can you try the commit ce93326 from [1.7_dev](https://github.com/lexus2k/ssd1306/tree/1.7_dev) branch?

RoboDurden, 2019-07-26:
Awesome :-)
I installed the `ssd1306-1.7_dev.zip` and it works perfectly. I see that all the changes of the `ce93326 ` commit are already implemented.

36 seconds: https://youtu.be/avvdXl-W4JU

Do you want me to test these `ce93326 ` changes in the version 1.7.20 i had installed via Arduino IDE ?

lexus2k, 2019-07-26:
You're welcome. :) I need to merge it to master branch and update library version. Arduino system publish new libraries in 24 hours.

lexus2k, 2019-07-26:
I published 1.7.21 version. It will take some time for new version to appear in Arduino IDE.

RoboDurden, 2019-07-26:
Yes, 1.7.21 is online and works perfectly :-)
Now i can implement your lib into my https://github.com/RoboDurden/openPanasonicBike
:-)

closed 2019-07-30 by RoboDurden