lexus2k/ssd1306 #89: i could make a font online compiler..

RoboDurden, 2019-07-26:
As a Windows user i put the fontgenerator.py onto my Ubuntu vServer.
And now could easily program a simple online compiler, where you upload a ttf or glcd file, set the ttf size ( the -SB option fails.) and copy and paste the c code.

But maybe someone has done that already :-)

```
def print_help_and_exit():
    print "Usage: ttf_fonts.py [args] > outputFile"
    print "args:"
    print "      --ttf S   use ttf name as source"
    print "      --glcd S  use glcd file as as source"
    print "      -s <N>    font size (this is not pixels!) "
    print "      -SB <N>   limit size in pixels to value (pixels will be cut)"
    print "      -fh       fixed height"
    print "      -fw       fixed width"
    print "      -g <S> <E> add chars group to the font"
    print "      -f old    old format 1.7.6 and below"
    print "      -f new    new format 1.7.8 and above"
    print "      -d        Print demo text to console"
    print "      --demo-only Prints demo text to console and exits"
    print "Examples:"
    print "   [convert ttf font to ssd1306 old format]"
    print "      ttf_fonts.py --ttf FreeSans.ttf -s 8 -f old > ssd1306font.h"
    print "   [convert ttf font to ssd1306 new format with demo text and print to console]"
    print "      ttf_fonts.py --ttf FreeSans.ttf -d -f new"
    print "   [convert GLCD font generated file to new format]"
    print "      ttf_fonts.py --glcd font.c -f new > ssd1306font.h"
    exit(1)
```

ideas welcome !

lexus2k, 2019-07-26:
Hi.

I fixed the issue with SB argument. Now it works. It would be nice to have online compiler for the fonts. It will be very useful.

> ./fontgenerator.py --ttf FreeSans.ttf -s 40 -SB 48 -f new -d > 1.txt

```.txt
// -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
// -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
// -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
// -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
// @@@@@----------------@@@@@@---------------@@@@@@------------------------------------------@@@@@---------------------@@@@-@@@@------------------------------------------------------------------
// @@@@@@---------------@@@@@@---------------@@@@@-------------------------------------------@@@@@---------------------@@@@-@@@@------------------------------------------------------------------
// @@@@@@--------------@@@@@@@---------------@@@@@-------------------------------------------@@@@@---------------------@@@@-@@@@-----------------------------------@@@@@@@--------------------@@@-
// -@@@@@--------------@@@@@@@---------------@@@@@-------------------------------------------@@@@@---------------------@@@@-@@@@---------------------------------@@@@@@@@@@@------------------@@@-
// -@@@@@--------------@@@@@@@@-------------@@@@@@-------------------------------------------@@@@@---------------------@@@@-@@@@-------------------------------@@@@@@@@@@@@@@@----------------@@@-
// -@@@@@@-------------@@@@@@@@-------------@@@@@--------------------------------------------@@@@@---------------------@@@@-@@@@------------------------------@@@@@@@@@@@@@@@@@--------------@@@@-
// -@@@@@@------------@@@@@@@@@-------------@@@@@--------------------------------------------@@@@@---------------------@@@@-@@@@------------------------------@@@@@@-----@@@@@@@-------------@@@@-
// --@@@@@------------@@@@@@@@@@------------@@@@@--------------------------------------------@@@@@---------------------@@@@-@@@@-----------------------------@@@@@@--------@@@@@------------@@@@@-
// --@@@@@------------@@@@-@@@@@-----------@@@@@@--------------------------------------------@@@@@---------------------@@@@-@@@@----------------------------@@@@@@---------@@@@@@----------@@@@@@-
// --@@@@@------------@@@@-@@@@@-----------@@@@@---------------------------------------------@@@@@---------------------@@@@-@@@@----------------------------@@@@@-----------@@@@@------@@@@@@@@@@-
// --@@@@@@----------@@@@@--@@@@-----------@@@@@------------@@@@@@@@-------------------@@@@@-@@@@@---------@@@@@@@-----@@@@-@@@@---------@@@@@@@------------@@@@@------------@@@@---@@@@@@@@@@@@@-
// ---@@@@@----------@@@@@--@@@@@----------@@@@@----------@@@@@@@@@@@@--------@@@@----@@@@@@-@@@@@-------@@@@@@@@@@@---@@@@-@@@@-------@@@@@@@@@@@----@@@--@@@@@-------------@@@@@--@@@@@@@@@@@@@-
// ---@@@@@----------@@@@---@@@@@----------@@@@@--------@@@@@@@@@@@@@@@@------@@@@--@@@@@@@@-@@@@@-----@@@@@@@@@@@@@@@-@@@@-@@@@-----@@@@@@@@@@@@@@@--@@@--@@@@@-------------@@@@@--@@@@@@@@@@@@@-
// ---@@@@@---------@@@@@----@@@@---------@@@@@--------@@@@@@@@@@@@@@@@@@-----@@@@--@@@@@@@@-@@@@@----@@@@@@@@@@@@@@@@@@@@@-@@@@----@@@@@@@@@@@@@@@@@-@@@--@@@@@-------------@@@@@-----------@@@@-
// ---@@@@@@--------@@@@@----@@@@---------@@@@@-------@@@@@@@------@@@@@@@----@@@@-@@@@@-----@@@@@---@@@@@@@@-----@@@@@@@@@-@@@@----@@@@@@@-----@@@@@@@@@--@@@@@--------------@@@@-----------@@@@-
// ----@@@@@--------@@@@@----@@@@@--------@@@@@-------@@@@@@---------@@@@@----@@@@@@@@-------@@@@@---@@@@@@---------@@@@@@@-@@@@---@@@@@@---------@@@@@@@--@@@@---------------@@@@@----------@@@@-
// ----@@@@@--------@@@@-----@@@@@--------@@@@@------@@@@@@-----------@@@@@---@@@@@@@--------@@@@@--@@@@@@-----------@@@@@@-@@@@--@@@@@@-----------@@@@@@-@@@@@---------------@@@@@----------@@@@-
// ----@@@@@-------@@@@@------@@@@-------@@@@@-------@@@@@------------@@@@@---@@@@@@---------@@@@@--@@@@@------------@@@@@@-@@@@--@@@@@------------@@@@@@-@@@@@---------------@@@@@----------@@@@-
// ----@@@@@@------@@@@@------@@@@@------@@@@@------@@@@@--------------@@@@@--@@@@@----------@@@@@-@@@@@@-------------@@@@@-@@@@--@@@@@-------------@@@@@-@@@@@---------------@@@@@----------@@@@-
// -----@@@@@------@@@@-------@@@@@------@@@@@------@@@@@--------------@@@@@--@@@@@----------@@@@@-@@@@@--------------@@@@@-@@@@-@@@@@--------------@@@@@-@@@@@---------------@@@@@----------@@@@-
// -----@@@@@------@@@@-------@@@@@------@@@@@------@@@@@---------------@@@@--@@@@@----------@@@@@-@@@@@---------------@@@@-@@@@-@@@@@--------------@@@@@-@@@@@---------------@@@@@----------@@@@-
// -----@@@@@-----@@@@@--------@@@@------@@@@-------@@@@----------------@@@@--@@@@@----------@@@@@-@@@@@---------------@@@@-@@@@-@@@@@---------------@@@@-@@@@@---------------@@@@@----------@@@@-
// ------@@@@-----@@@@@--------@@@@@----@@@@@-------@@@@----------------@@@@@-@@@@@----------@@@@@-@@@@----------------@@@@-@@@@-@@@@----------------@@@@-@@@@@---------------@@@@@----------@@@@-
[... cut ...]
// --------------------------------------------------------------------------------------------------------------------------------------------------@@@@-----------------------------------------
// --------------------------------------------------------------------------------------------------------------------------------------------------@@@@-----------------------------------------
// --------------------------------------------------------------------------------------------------------------------------------------------------@@@@-----------------------------------------
```

RoboDurden, 2019-07-27:
okay here is my online compiler: www.pionierland.de/fonts

In which branch do i find the updated fontgenerator.py ??? I only see 10 months old versions !

Should i add the -fh and -fw options ? Don't really know what they do.

ideas welcome :-)

lexus2k, 2019-07-27:
It is master branch. Fontgenerator.py uses modules in folder tools/modules. The bug location was inside fontcontainer.py. As for -fh and -fw options, they can be used to create fonts of fixed height or fixed width. Fixed width option makes all chars of the same width, regardless of their content (`! # A` will have the same width). The same behavior for fixed height option.
I added the link to wiki pages. Thank you a lot.

RoboDurden, 2019-07-27:
Yes i downloaded the master branch and have activated the -SB and -fw and -fh options :-)
Also added asc-charts for the public fonts (that i quickly downloaded from some public-domain-download-site):

www.pionierland.de/fonts

I think that's it for now. But ideas always welcome.
Am happy that you like my online compiler :-))

Buka666, 2019-11-19:
Hello. Everything is great. But there is no way to use an encoding greater than 127. Could you add this feature.

RoboDurden, 2019-11-21:
Yes i restrict (clamp) the parameters:
`	Clamp($h,'iSize',6,100);
	Clamp($h,'iHeight',6,128);
	Clamp($h,'iFrom',32,127);
	Clamp($h,'iTo',32,127);`

What max iSize,iHeight do you wish for ?

Buka666, 2019-11-21:
Need Cyrillic encoding. From 1040 to 1103.

RoboDurden, 2019-11-21:
`	Clamp($h,'iSize',6,250);
	Clamp($h,'iHeight',6,256);
	Clamp($h,'iFrom',32,127);
	Clamp($h,'iTo',32,65536);`

:-)

Buka666, 2019-11-21:
Thank you so much. You made my day beautiful.

RoboDurden, 2019-11-21:
:)

lexus2k, 2019-11-24:
@RoboDurden ,

I'm glad that you solved the problem
Thank you so much

RoboDurden, 2022-01-28:
Hello again.
I am moving my online font compiler to a new ubuntu server. As i strongly detest Linux i have already forgotton what was neccessary to run your fontgenerator.py :-/ 
I have Python 2.7.12 running but  `python fontgenerator.py --ttf public/Avocado.ttf -s 20 -SB 24 -g 32 95 -f new -d > ready/2076019679.h` outputs an empty file.

Your install comments does not make much sense to me:
```
# Linux:
#     Fixing pip
#     curl https://bootstrap.pypa.io/get-pip.py | python2
#     sudo pip install freetype-py

```

Fixing pip of course returns `-bash: Fixing: command not found`

Will continue on my own, but maybe someone would like to tell me all the easy answers :-)

RoboDurden, 2022-01-29:
Okay it seems that i had not python2.7 installed and
```
apt install pip
pip install freetype-py
```
did install the module only for python3

This somehow worked:
```
python --version
apt install python
which pip2.7
wget https://bootstrap.pypa.io/pip/2.7/get-pip.py
sudo python2.7 get-pip.py
which pip2.7
pip2.7 install freetype-py
```

Silly full-control-commandline-Linux believers who still live in the 90s of last century..
For the next few days, the font compiler might be here instead: https://robodurden.de/fonts/

RoboDurden, 2022-01-29:
Okay i installed your latest python code that will also run on python3.
But now i see that your ouptut is different:

`python fontgenerator.py --ttf public/Avocado.ttf -s 20 -SB 24 -g 32 2 -f new -d > ready/4361083041.h`

[0, 0]
[6, 0, 12, 0, 8, 0, 24, 0, 24, 0, 48, 0, 48, 0, 48, 0, 112, 0, 96, 0, 96, 0, 96, 0, 64, 0, 0, 0, 0, 0, 96, 0, 192, 0, 192, 0, 128, 0]
[18, 0, 62, 0, 60, 0, 108, 0, 104, 0, 88, 0, 160, 0]
// --------
// --------
// -----@@-
// ----@@--
// ----@---
// ---@@---
// ---@@---
// --@@----
// --@@----
// --@@----
// -@@@----
// -@@-----
// -@@-----
// -@@-----
// -@------
// --------
// --------
// -@@-----
// @@------
// @@------
// @-------
// --------
// --------
// --------
extern const uint8_t free_Avocado7x24[] PROGMEM;
const uint8_t free_Avocado7x24[] PROGMEM =
{
//  type|width|height|first char
    0x02, 0x07, 0x18, 0x00,
// GROUP first ' ' total 3 chars
//  unicode(LSB,MSB)|count
    0x00, 0x20, 0x03, // unicode record
    0x00, 0x00, 0x01, 0x00,// char ' ' (0x0020/32)
    0x00, 0x00, 0x07, 0x15,// char '!' (0x0021/33)
    0x00, 0x15, 0x07, 0x07,// char '"' (0x0022/34)
    0x00, 0x1C,
    // char ' ' (0x0020/32)
    0x00, 0x00, 0x80, 0xE0, 0x78, 0x0C, 0x04, 0x00, 0x7C, 0x3F, 0x07, 0x00, 0x00, 0x00, 0x1C, 0x0E, 0x02, 0x00, 0x00, 0x00, 0x00, // char '!' (0x0021/33)
    0x40, 0x38, 0x5E, 0x27, 0x3E, 0x0E, 0x03, // char '"' (0x0022/34)
    0x00, 0x00, 0x00, // end of unicode tables
    // FONT REQUIRES 52 BYTES
};

python fontgeneratorOld.py --ttf public/Avocado.ttf -s 20 -SB 24 -g 32 2 -f new -d > ready/1208376149.h

```
// --------
// --------
// -----@@-
// ----@@--
// ----@---
// ---@@---
// ---@@---
// --@@----
// --@@----
// --@@----
// -@@@----
// -@@-----
// -@@-----
// -@@-----
// -@------
// --------
// --------
// -@@-----
// @@------
// @@------
// @-------
extern const uint8_t free_Avocado7x21[] PROGMEM;
const uint8_t free_Avocado7x21[] PROGMEM =
{
//  type|width|height|first char
    0x02, 0x07, 0x15, 0x00,
// GROUP first ' ' total 3 chars
//  unicode(LSB,MSB)|count
    0x00, 0x20, 0x03, // unicode record
    0x00, 0x00, 0x01, 0x00, // char ' ' (0x0020/32)
    0x00, 0x00, 0x07, 0x15, // char '!' (0x0021/33)
    0x00, 0x15, 0x07, 0x07, // char '"' (0x0022/34)
    0x00, 0x1C,
    // char ' ' (0x0020/32)
    0x00, 0x00, 0x80, 0xE0, 0x78, 0x0C, 0x04, 0x00, 0x7C, 0x3F, 0x07, 0x00, 0x00, 0x00, 0x1C, 0x0E, 0x02, 0x00, 0x00, 0x00, 0x00, // char '!' (0x0021/33)
    0x40, 0x38, 0x5E, 0x27, 0x3E, 0x0E, 0x03, // char '"' (0x0022/34)
    0x00, 0x00, 0x00, // end of unicode tables
    // FONT REQUIRES 52 BYTES
};

```

It is not the fontgenerator.py itself but one of these three modules:

```
from modulesOld import ttfsource
from modulesOld import glcdsource
from modulesOld import fontgenerator

```

I added a button "old modules" to my compiler but that will only confuse users.
Don't know right now if i anyone uses my compliler.

lexus2k, 2022-01-29:
Hello

> Okay it seems that i had not python2.7 installed and

The script works for both Python 2.7 and Python 3. At least on the system I have (Ubuntu 20.04).
The Python3 doesn't require all those workaround steps, so it is enough to install only freetypes library.

> For the next few days, the font compiler might be here instead: https://robodurden.de/fonts/

That would be great.

> [0, 0]

Yeah, you can take the script from lcdgfx library for now as it doesn't have this debug output.
P.S. I fixed this issue in this library. See the commit above.

RoboDurden, 2022-01-29:
Hi Lexus, keep in mind that i at the moment have no idea what your libraries do as i have not worked with it for some time.
For my onliine compiler i use the code from https://github.com/lexus2k/ssd1306
Should i replace all code with https://github.com/lexus2k/lcdgfx/tree/master/tools ?
Or should i offer a compiler option to choose whether to take the python files from lcdgfx or ssd1306 ?

RoboDurden, 2022-01-29:
Okay i was confuse by this debug output. Have removed `print(bitmap.buffer)` manually as it has not yet shown up publicy.
Say so if the code is newer on lcdgfx, than i will copy the scripts from there :-)