lexus2k/ssd1306 #92: Communication with other I2C devices (ATTiny84)

tom-4, 2019-09-07:
Hello,
I have a similar problem like in issue #48 , but, ATTiny84 this time not ATTiny, so I read your solution (commit: 384dcee).
I just copied the whole text and pasted it to the same io file. I just changed X5 with X4  and it works :)

lexus2k, 2019-09-09:
Hi Tom,

I will try to fix as soon as possible and let you know :)

lexus2k, 2019-09-10:
Hi Tom,

Can you check that code from [1.7_dev](https://github.com/lexus2k/ssd1306/tree/1.7_dev) branch works for you?
It would be nice if you let me know the board you use in Arduino IDE and package installed for that board

tom-4, 2019-09-10:
Hi there,
It works like a charm! Thank you.
I used Arduino nano (clone) as a programmer. Arduino IDE 1.8.9.
Board cores for ATTiny84 are ATTinycore. https://github.com/SpenceKonde/ATTinyCore
Regards,
Tommy

lexus2k, 2019-09-11:
Ok, thank you for report

closed 2019-09-11 by lexus2k