lexus2k/ssd1306 #95: AtTiny84 not working

programer786, 2019-12-02:
Hello,

First of all, thanks for this library, I really like it a lot.

I am trying to use the library for a project that uses an AtTiny84A and a 128x32 SSD1306 OLED. I'm using the I2C version.

The OLED stays completely black though, no matter what I do.
I have tested the included Demo and it works with an AtTiny85 but the screen remains black with the AtTiny84A.
I have also tried the code from issue #39 but the result is still the same.

I have tried both the ATTinyCore (with clockwise pin mapping) and the core by D. Amellis.
I'm using the internal 8MHz clock and an Arduino Nano as ISP programmer.
I have connected SDA to pin 7 and SCL to pin 9 (software pins 6 and 4 respectively).

lexus2k, 2019-12-03:
Hi @programer786,

I'm looking into issue. Maybe, it relates to i2c timings. I will let you know, if I find something.

Best regards

lexus2k, 2019-12-08:
Hi,

I've checked the code (i2c software implementation), and didn't find any problems. Do you have any logic analyzer to capture the data on 4 and 6 logic pins?

lexus2k, 2019-12-22:
Hi,

Can you check [1.8_dev](https://github.com/lexus2k/ssd1306/tree/1.8_dev) branch?

lexus2k, 2019-12-26:
Did you have a chance to verify the fix?

lexus2k, 2020-01-01:
Closing the issue due to inactivity.

closed 2020-01-01 by lexus2k