lexus2k/ssd1306 #96: Numbers

classic-audio, 2019-12-31:
I cannot really get this working:
Voltage = analogRead(A0);

How do I print Voltage on an OLED display?
(I have looked very much for an example)

It is a matter of converting the number to a string - how to do that correct for this library?

lexus2k, 2020-01-01:
Hello again. :)
Happy new year.

BTW, there is not dedicated function to print numbers, but you can do the following:
```.cpp
#include "ssd1306.h"
#include <stdio.h>

void printNumber(int x, int y, int number)
{
    char intStr[12];
    snprintf(intStr, sizeof(intStr), "%i", number );
    ssd1306_printFixed (x, y, intStr, STYLE_NORMAL);
}
```

classic-audio, 2020-01-02:
Hello Alex,
Thank you very much for the fast reply and help.
-----------------------------------------------------
Your remark:
Yes actually I have an idea or wish :)

I am using your library for an application in audio test equipment.
My plan is:
To use ATtiny85 as Master (I2C) and drivng a display OLED color SSD1331.
The display show show the measuring results from 2 slaves - ATtiny85's

The two slave a place close to the source of measuring.
The display with ATtiny84 Master at the front panel-
Communication: I2C
This is a technical great solution - avoiding long sensible signal cables - replaced with I2C bus and a master / slave system.

But as far as I can read - at forums
It is not possible for ATtiny84 to drive 
1. Master / slave communication - I2C
and
2. OLED SSD1331 - SPI 
at the same time?

What is your opinion ?

lexus2k, 2020-01-02:
Well, the answer could be to implement software i2c bus for attiny85.
Attiny85 still has 5 pins for use: 3 pins (PB0, PB1, PB2) can be used as SPI to control ssd1331 display (via USI hardware), and PB3, PB4 can be used as i2c master to hold communication with 2 i2c slaves. software i2c master implementation of ssd1306 library supports only writing to slave devices, but it can be extended to be used for reading from slaves.

attiny84 also has USI hardware to implement SPI bus (PA4, PA5, PA6), and for I2C communication you have to use software version.
I think that it is possible to implement what you need on both microcontrollers: Attiny84 and Attiny85.

But what about ESP32? It is more powerful, you will get wifi-interface for your final devices, and will implement both i2c and spi communication without any troubles. And it is still quit cheap.
In case you build pcb by yourself, it is also easy to solder esp32 wroom modules manually.

classic-audio, 2020-01-02:
Thank you very much.
This means - there is a solution.
I will be going for the ATtiny84 solution - if possible.

But..........
I dont have the knowledge of how to implement a software i2c solution on ATtiny84.
I am sorry to ask - but do you have a link og an example (better of coarse for me :) )

lexus2k, 2020-01-03:
If you're ok with C++, then I did some changes to software i2c in another my library. So, it will allow to read from slave devices using software implementation. But I'm not sure if it works - I don't have i2c responding on read command slaves by hand right now. Many i2c displays are usually write only devices.
https://github.com/lexus2k/lcdgfx/blob/v1.0.0_dev/src/lcd_hal/avr/ssd1306_i2c_embedded.cpp
https://github.com/lexus2k/lcdgfx/blob/v1.0.0_dev/src/lcd_hal/avr/ssd1306_i2c_embedded.h

If you test this class, it would be nice.
And again, I would recommend esp32. It is really cool microchip

lexus2k, 2020-01-04:
It's me again. Finally I tested software implementation, found 1 bug, and fixed it on dev branch.
https://github.com/lexus2k/lcdgfx/blob/v1.0.0_dev/src/lcd_hal/avr/ssd1306_i2c_embedded.cpp
https://github.com/lexus2k/lcdgfx/blob/v1.0.0_dev/src/lcd_hal/avr/ssd1306_i2c_embedded.h

If you want to test, you can download lcdgfx library from my v1.0.0_dev branch, add it to your Arduino IDE, include "lcdgfx.h" header, and try software implementation for Attiny85 or Attiny84 (whichever you prefer).

Here are my sketch I used for testing with DS3231 RTC chip:
```.cpp
#include "lcdgfx.h"

SoftwareI2c i2c(-1, -1, 0);

void initDs3231()
{
    i2c.start(0x68, false);
    i2c.send( 0x0E );
    i2c.send( 0B00011100 );
    i2c.send( 0B00110000 );
    i2c.stop();
}

void setup()
{
    i2c.begin();
    Serial.begin(115200);
    initDs3231();
}

void readDs3231()
{
    // Open i2c in write mode to set register number for DS3231
    i2c.start(0x68, false);
    i2c.send( 0x00 );
    i2c.stop();

    // open i2c in read mode to read from ds3231 registers
    i2c.start(0x68, true);
    uint8_t seconds = i2c.receive( false );
    uint8_t minutes = i2c.receive( true ); // true indicates last byte to read
    i2c.stop();
    Serial.print( (minutes & 0x0F) + (minutes>>4) * 10 );
    Serial.print( ':' );
    Serial.println( (seconds & 0x0F) + (seconds>>4) * 10 );
}

void loop()
{
    readDs3231();
    lcd_delay(1000);
}
```

The sketch demonstrates, how i2c software implementation can be used for reading from slaves. You will see in Arduino IDE monitor RTC clocks output ticking each second. I think you can easily figure out, how to use `SoftwareI2c`

lexus2k, 2020-01-10:
Hi Frede,

did you have a chance to check example above?

lexus2k, 2020-01-25:
Any updates on spi?

closed 2020-03-05 by lexus2k