lexus2k/ssd1306 #97: Screen shifted 1 row and 1 column

MaxScorda, 2020-01-03:
Hi,
with a il9163 128x128 display, library works like a charm but theres'a a problem.
Last row and last column are full of garbage and first row is row 1 (not 0), the same for the columns.

lexus2k, 2020-01-03:
Hi

Can you take some pictures for me?
If you have simple sketch to reproduce the issue, it will be very helpful.

MaxScorda, 2020-01-03:
Thank you for your answer,
Here a picture where you can see last row and last column full of garbage.
Row 0 and col 0 are off screen and first visible row is 1 and the same for column
This is init function
void InitScreen() {
  ssd1306_setFixedFont(ssd1306xled_font6x8);
  il9163_128x128_spi_init(12, 10, 9);
  ssd1306_setMode( LCD_MODE_NORMAL );
  ssd1306_clearScreen8( );
}
but later i can post a full  sample sketch

[image]

Thank you!

lexus2k, 2020-01-04:
If you put pixel at 0,0 position, do you see it on the screen?
Are you sure that you have 128x128 display (Maybe it is 132x132)?
I'm asking, because after reading ili9163 datasheet, I didn't find any offset command. Of course, it is possible to fix your problem by adding fixed offset to lcd_il9163.c source code. But we need to find root cause for the problem first.

MaxScorda, 2020-01-05:
No I can't see 0,0
This is a demo code
`#include "ssd1306.h"
void setup() {
  ssd1306_setFixedFont(ssd1306xled_font6x8);
  il9163_128x128_spi_init(12, 10, 9);
  ssd1306_setMode( LCD_MODE_NORMAL );
  ssd1306_clearScreen8( );

  ssd1306_setColor(RGB_COLOR8(255, 255, 255));
  ssd1306_printFixed8(0 , 0, "ABCD" , STYLE_BOLD);
  ssd1306_printFixed8(122 , 0, "A" , STYLE_BOLD);
  ssd1306_printFixed8(0 , 20, "0234" , STYLE_BOLD);
  ssd1306_printFixed8(1 , 50, "0234" , STYLE_BOLD);

  ssd1306_drawRect8(0, 0, ssd1306_displayWidth() - 1, ssd1306_displayHeight() - 1);
}
`
[image]

and you can see only right and bottom square border
It could be a defected display? I haven't onother one to test...
Thank you

lexus2k, 2020-01-06:
Hi Max,

I added new function `il9163_setOffset`, use it after display initializaion.
Can you check if the changes work for your (https://github.com/lexus2k/ssd1306/tree/1.8_dev).
Do you have any link to the store, selling the displays like yours?
I'm wondering if they built on top of modified controller.

MaxScorda, 2020-01-06:
Thank you, it's perfect now (I also discover x-offset was 2 pixel off il9163_setOffset(2, 1))
Sorry but i bought display some years ago and never used since this weeks and Ebay don't show orders older than 3 years.
However I've ordered another one and let me see how it works.
Thank you so much!

lexus2k, 2020-01-08:
You're welcome.
Let me know if you find some other issues.

closed 2020-01-16 by MaxScorda