lexus2k/ssd1306 #98: Only fonts starting ssd1306xled work

rickyelqasem, 2020-01-10:
**Describe the bug**
If choose any other font than the ones starting ssd1306xled I get corruption on the screen. ssd1306xled fonts and digital_font5x7 work fine. Is there something special I need to do?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
All fonts work

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**
 - 1.8.1
 - OLED 128x64
 - Windows
 - Atmega328p
 - PlatformIO

**Additional context**
Add any other context about the problem here.

lexus2k, 2020-01-11:
Hi @rickyelqasem 

Could you please provide more details? Like sketch code, which doesn't work.
BTW, there are 2 types of fonts, available in the library: Fixed width fonts, and Free type fonts. To set the font, you need to use different functions, depending on selected font type: `ssd1306_setFixedFont` or `ssd1306_setFreeFont`.

rickyelqasem, 2020-01-11:
Ok thanks.. Is there any more fonts you know of?

lexus2k, 2020-01-12:
You can create your own fonts for the library. Please, read this instruction: https://github.com/lexus2k/ssd1306/wiki/How-to-create-new-font-for-the-library

closed 2020-01-25 by lexus2k